xoseperez/espurna #1245: MQTT messages are sent regardless of the bound action

MHalter, 2018-10-05:
I have created my own hardware definition for a NodeMCU to use my own button 2.

`#elif defined(NODEMCU_DEVKIT)

    // Info
    #define MANUFACTURER        "NODEMCU"
    #define DEVICE              "DEVKIT"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY           1

    #define BUTTON2_PIN         14
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON2_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON2_LNGCLICK    BUTTON_MODE_NONE
    #define BUTTON2_LNGLNGCLICK BUTTON_MODE_NONE

    // Hidden button will enter AP mode if dblclick and reset the device when long-long-clicked
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // Light
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1`

 It compiles and runs so far with platformio on Visual Studio Code

As you could see I set the events BUTTON2_DBLCLICK etc. to BUTTON_MODE_NONE. I did that in the belief that these events will not be transmitted to a MQTT. However, I still see the values 3, 4 and 5 in the MQTT.

[image]

[image]

. 

Found on
Version : ESPURNA 1.13.3a
SDK : 1.5.3(aec24ac9)
Core : 2.3.0

Thanks

Mark

Valcob, 2018-10-05:
Made a fix for it, it will allow either to send messages only for events  with bound actions or all events regardless of bindings:

>     unsigned char action = buttonAction(id, event);
> 
>     #if MQTT_SUPPORT
>         if (action != BUTTON_MODE_NONE || BUTTON_MQTT_SEND_ALL_EVENTS)
>             buttonMQTT(id, event);
>     #endif 

will create a PR in couple of hours

xoseperez, 2018-10-07:
PR #1246 merged. Thank you!

closed 2018-10-07 by xoseperez