xoseperez/espurna #1248: With HLW Sometimes program crashes.

arihantdaga, 2018-10-06:
I checked with tinkerman_espurna_h06 board's code. 

Sometimes ESP Crashes in power measurement code with Exception 29. 

I have decoded error stack 

```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x4000e1c3
EXCVADDR: 0x00000018

Decoding stack results
0x4010070c: pvPortMalloc at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.c line 153
0x40100cec: malloc at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1672
0x4010070c: pvPortMalloc at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.c line 153
0x401001ea: HLW8012::cf1_interrupt() at /Users/arihantdaga/Documents/Arduino/libraries/xoseperez-hlw8012-cc3baec18f87/src/HLW8012.cpp line 189
0x401001ea: HLW8012::cf1_interrupt() at /Users/arihantdaga/Documents/Arduino/libraries/xoseperez-hlw8012-cc3baec18f87/src/HLW8012.cpp line 189
0x40100720: pvPortZalloc at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.c line 174
0x40100d20: calloc at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1699
0x401004a8: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 130
0x4022b7df: loop_task(ETSEvent*) at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 133
0x401001ea: HLW8012::cf1_interrupt() at /Users/arihantdaga/Documents/Arduino/libraries/xoseperez-hlw8012-cc3baec18f87/src/HLW8012.cpp line 189
0x401000cc: _hlw8012_sensor_isr(unsigned char) at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 287
0x40100160: _hlw8012_sensor_isr_12() at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 295
0x40100579: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 166
0x40100568: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 162
0x401004a8: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 130
0x401001ea: HLW8012::cf1_interrupt() at /Users/arihantdaga/Documents/Arduino/libraries/xoseperez-hlw8012-cc3baec18f87/src/HLW8012.cpp line 189
0x401000cc: _hlw8012_sensor_isr(unsigned char) at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 287
0x40100160: _hlw8012_sensor_isr_12() at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 295
0x40100579: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 166
0x40100568: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 162
0x401001ea: HLW8012::cf1_interrupt() at /Users/arihantdaga/Documents/Arduino/libraries/xoseperez-hlw8012-cc3baec18f87/src/HLW8012.cpp line 189
0x401000cc: _hlw8012_sensor_isr(unsigned char) at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 287
0x40100160: _hlw8012_sensor_isr_12() at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 295
0x40100579: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 166
0x40100568: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 162
0x401001ea: HLW8012::cf1_interrupt() at /Users/arihantdaga/Documents/Arduino/libraries/xoseperez-hlw8012-cc3baec18f87/src/HLW8012.cpp line 189
0x401001ea: HLW8012::cf1_interrupt() at /Users/arihantdaga/Documents/Arduino/libraries/xoseperez-hlw8012-cc3baec18f87/src/HLW8012.cpp line 189
0x401000cc: _hlw8012_sensor_isr(unsigned char) at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 287
0x4022f72c: _printf_i at ../../../.././newlib/libc/stdio/nano-vfprintf_i.c line 249
0x40100160: _hlw8012_sensor_isr_12() at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 295
0x40100579: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 166
0x40100568: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 162
0x401004a8: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 130
0x4023391d: __ssputs_r at ../../../.././newlib/libc/stdio/nano-vfprintf.c line 240
0x4022f32d: _printf_common at ../../../.././newlib/libc/stdio/nano-vfprintf_i.c line 94
0x401001ea: HLW8012::cf1_interrupt() at /Users/arihantdaga/Documents/Arduino/libraries/xoseperez-hlw8012-cc3baec18f87/src/HLW8012.cpp line 189
0x401000cc: _hlw8012_sensor_isr(unsigned char) at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 287
0x40100160: _hlw8012_sensor_isr_12() at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 295
0x40100579: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 166
0x40100568: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 162
0x401004a8: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 130
0x40235178: memchr at ../../../.././newlib/libc/string/memchr.c line 126
0x40233c20: _svfprintf_r at ../../../.././newlib/libc/stdio/nano-vfprintf.c line 612
0x401001ea: HLW8012::cf1_interrupt() at /Users/arihantdaga/Documents/Arduino/libraries/xoseperez-hlw8012-cc3baec18f87/src/HLW8012.cpp line 189
0x401000cc: _hlw8012_sensor_isr(unsigned char) at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 287
0x40100160: _hlw8012_sensor_isr_12() at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 295
0x40100579: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 166
0x40100568: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 162
0x40233830: __ssputs_r at ../../../.././newlib/libc/stdio/nano-vfprintf.c line 180
0x401001ea: HLW8012::cf1_interrupt() at /Users/arihantdaga/Documents/Arduino/libraries/xoseperez-hlw8012-cc3baec18f87/src/HLW8012.cpp line 189
0x401000cc: _hlw8012_sensor_isr(unsigned char) at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 287
0x40217d49: Embedis::kvs_work(Embedis::Dictionary*, char const*, int, char const*, unsigned int*, unsigned int*) at /Users/arihantdaga/Documents/Arduino/libraries/Embedis/src/Embedis.cpp line 1097
0x40100160: _hlw8012_sensor_isr_12() at /var/folders/j5/gn2d33zn7vb8k87m0vtm9th00000gn/T/arduino_build_350190/sketch/config/../sensors/HLW8012Sensor.h line 295
0x40100579: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 166
0x40100568: interrupt_handler at /Users/arihantdaga/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_digital.c line 162
0x40203ae0: __lambda55::_FUN(size_t) at /Users/arihantdaga/Documents/Arduino/Espurna/espurna/code/espurna/settings.ino line 461
[... cut ...]
[300597] [MAIN] Heap  : 41248 bytes total - 16064 bytes used (38) - 25184 bytes free (61)
[300598] [MAIN] Power: 3029 mV
```

I am using a custom board with hlw and in my case hlw pins are  @ GPIO(13,14,12). 
it works fine for sometimes, but sometimes randomly it crashes with the above stack. 
- my hardware is a simple relay and an hlw,
- i am using latest code from espurna
- i am using latest esp core as well
- I only modified hlw pin numbers from hardware.h file to 13,14,12 for tinkerman_espurna_h06  and uncommented tinkerman_espurna_h06 in arduino.h.

xoseperez, 2018-10-07:
That is a pretty long error stack and look kind of recursive... Maybe the device runs out of stack space but I'm not sure... Have you tried using Arduino Core 2.3.0? I know it's "old" but it's the best-supported version by ESPurna.

arihantdaga, 2018-10-07:
Ohh.. Noh i did not try the same with 2.3.0. But I'll check and confirm.

xoseperez, 2018-12-13:
I'm closing this now. I have added a comment on the HLW8012 section in the sensors page about this. If it happens again, feel free to reopen with new data.

closed 2018-12-13 by xoseperez