xoseperez/espurna #1251: Sonoff TH - How to change the GPIO from 14 to 4?

lev2k, 2018-10-07:
Or even better, add GPIO 4, so I can use GPIO 14 AND 4 together.
Atm, I'm using a pre-compiled binary from here and it seems that only GPIO 14 is hardcoded.
Is it possible to add/switch the GPIO via Console/Telnet, or do I have to compile my own custom firmware?
Thanks in advance

xoseperez, 2018-10-07:
At the moment you have to compile it yourself. This will change with v2. You will also have to modify the hardware to wire GPIO4 too. Check this post here: http://tinkerman.cat/sonoff-th10-th16-sensors-displays-actuators

lev2k, 2018-10-07:
well, I've compiled the firmware already, but I'm not sure, if I used the correct Syntax in "hardware.h".
When compiling in platform.io, I get the following warning, but at least, it seems to work..:

> In file included from espurna\config/all.h:29:0,
> from C:/Users/rol/Downloads/_SONOFF-TASMOTA_/ESPURNA/source/espurna-dev/code/espurna/espurna.ino:22:
> espurna\config/hardware.h:330:0: warning: "DHT_PIN" redefined [enabled by default]
> #define DHT_PIN             4
> ^
> espurna\config/hardware.h:329:0: note: this is the location of the previous definition
> #define DHT_PIN             14
> ^

hardware.h:
```
    #ifndef DHT_SUPPORT
    #define DHT_SUPPORT         1
    #endif
    #define DHT_PIN             14
    #define DHT_PIN             4
```

closed 2018-10-07 by lev2k