xoseperez/espurna #1252: OSError: [Errno 98] Address already in use

Crashmailman, 2018-10-07:
Hi all i am trying to a OTA and everytime i doit i am getting a error

OSError: [Errno 98] Address already in use

can someone please tell me where i am going wrong

xoseperez, 2018-10-07:
Can you provide here the way you are doing the OTA upgrade?

Crashmailman, 2018-10-07:
HI there i am just following the steps on the site
*Adam R Wolarczuk*
ADBY IT AND MEDIA SERVICES
+61 2 82946100 <+61+2+82946100> | 0405 041 601 | adam@adby.com.au |
www.adby.com.au | Skype: Wolarczuk <https://webapp.wisestamp.com/#> |
 Jordan Springs NSW
<http://facebook.com/ADBYITS>  <http://twitter.com/adbyits>

This e-mail message may contain confidential or legally privileged
information and is intended only for the use of the intended recipient(s).
Any unauthorized disclosure, dissemination, distribution, copying or the
taking of any action in reliance on the information herein is prohibited.
E-mails are not secure and cannot be guaranteed to be error free as they
can be intercepted, amended, or contain viruses. Anyone who communicates
with us by e-mail is deemed to have accepted these risks. Company Name is
not responsible for errors or omissions in this message and denies any
responsibility for any damage arising from the use of e-mail. Any opinion
and other statement contained in this message and any attachment are solely
those of the author and do not necessarily represent those of the company.

closed 2018-12-13 by stale[bot]