xoseperez/espurna #1254: 1.13.3a

AlbrtoSam, 2018-10-08:
I have just upgraded to 1.13.3a my dual sonoff R2 and when i reboot switch 1 gets ON. I have set the option boot mode as Always off

AlbrtoSam, 2018-10-08:
I also have a 4ch and it works fine on 13.3a

berryelectronics, 2018-10-21:
Noticed this too. I have freshly flashed Dual R2's here, and the first boot has also the Swith #1 turn on upon boot, aswell after setting them to Boot into Off

mcspr, 2018-11-28:
Mixed up with the #1216, cc @macobra as this is the actual R2 issue. 

Could this be related to button setup?
https://github.com/xoseperez/espurna/blob/e2641b9aed5e50e2502022fb0c029c4fdef09252/code/espurna/config/hardware.h#L490-L501
*edit:* i.e should BUTTON2 (GPIO9) be BUTTON_PUSHBUTTON as BUTTON3 (GPIO10)?

I am not that familiar with pin initialization requirements, but reading DebounceEvent source  BUTTON_SWITCH initial reading is done without pinMode setting it INPUT_PULLUP (or INPUT at all):
https://github.com/xoseperez/debounceevent/blob/4681547f72b226a9bafc95cf7a4623ec69f6a327/src/DebounceEvent.cpp#L36-L65

Core itself skips this too
(2.3.0 version linked, same with 2.4.X. My assumption that is because generic ESP8266 connects GPIO9/10 to SPI flash, unlike ESP8285 that is used in Dual R2):
https://github.com/esp8266/Arduino/blob/09826c6d87e54c15422da90f060377c9ef188947/cores/esp8266/core_esp8266_wiring_digital.c#L160-L176

Misiu, 2018-12-13:
@mcspr Is this fixable? If Yes, will this be fixed? I'm planning to buy a couple of those Sonoffs for roller shutters and I just found this issue.

mcspr, 2018-12-13:
My theory is `pinMode(..., INPUT)` should be done ([IDE does this for esp8285 board](https://github.com/esp8266/Arduino/blob/master/variants/esp8285/initD9D10Pins.cpp))

I do not have Dual R2, but testing on the similar esp8285 with rfbridge: `gpio 9` and `gpio 10` in terminal show `LOW`, but must read `HIGH`, which is why I wondered about the pin setup.

ralias, 2018-12-20:
I tried something. When I choose in the configuration SWITCH 1 Boot Mode: "Always ON", then it boot with the switch OFF.

ralias, 2018-12-23:
@macobra I try again paying more attention to the boot process. When I set Switch 1 Boot Mode: "Always ON", I reboot the device, first of all it put the switch 1 on, finally it apply the configuration option "Always ON" and it power off the switch. There are a up/down state. It's not only change the label "OFF" by "ON". My previous message It's a bypass to avoid a manual intervention after a unexpected reboot. It's not the solution.

xoseperez, 2018-12-27:
@ralias do you mean 1.13.2 works but 1.13.3 doesn't?

ralias, 2018-12-31:
> @ralias do you mean 1.13.2 works but 1.13.3 doesn't?

@xoseperez, exactly the 1.13.3 binary doesn't works and binary version 1.13.2 works as expected.

mileperhour, 2019-01-25:
FYI: The same issue, using 1.13.3, on Sonoff Bridge RF v2 hardware. It can't be fixed using "same as before", "always on" or "always off". All options send one order to every switch programmed on each Espurna boot. (I expected that "same as before" won't do any kind of order to switches on boot -¿?-). Its really difficult to handle using devices with same "on" and "off" RF code, as it can't be configured to be predictable on any situation.

xoseperez, 2019-02-10:
Can any one test the version in the dev branch after the DebounceEvent library update?

ColinShorts, 2019-02-11:
I tested this at 5f288fc2 on the dev branch on one device and can't reproduce this, my network is being a pain or I'd have tried on a few devices. If you need more testing let me know and I'll power cycle some.

skyynet, 2019-02-16:
Works for me. I just downloaded the current dev and upgraded Debounce Event Lib from 2.0.4 to 2.0.5 on my Sonoff Dual R2.

closed 2019-02-21 by xoseperez