xoseperez/espurna #1255: Question i2c: how to write to i2c

lblabr, 2018-10-08:
i'd like to use the sht3xi2c as template for connecting an DAC to my esp.

i have a question: how do i write this to i2c, as i have seen, i can write once or a buffer....

 Wire.beginTransmission(MCP4725_ADDR);
  Wire.write(64);                     // cmd to update the DAC
  Wire.write(sintab2[lookup] >> 4);        // the 8 most significant bits...
  Wire.write((sintab2[lookup] & 15) << 4); // the 4 least significant bits...
  Wire.endTransmission();

how do i achive this ?

thanks in advance

xoseperez, 2018-10-09:
Using the methods in the I2C module (assumint sintab2 is a uint16_t array):

```
i2c_write_uint16(MCP4725_ADDR, 64, sintab2[lookup] << 4);
```

lblabr, 2018-10-09:
what about the least significant byte for the command ?
i have seen, that there is always an new transmission on i2c_write_uint16

lblabr, 2018-10-09:
after thinking a while, i think i got it and will try, thanks for support!!

lblabr, 2018-10-10:
works ...

closed 2018-10-10 by lblabr