xoseperez/espurna #1256: Electrodragon relays not working after update to 1.13.3

skyynet, 2018-10-08:
I did an update from 1.13.2 to 1.13.3. Afterwards switching my relays vie HTTP Get did not work anymore.

I set Restful API to off, saved and rebooted. The Restful API setting in the GUI was again set to on.

Switching relays neither works via GET commands nor the ESPurna GUI.

mcspr, 2018-10-09:
I couldn't reproduce this. Saving from web did work and after rebooting `apiRestFul` setting was still there.
Does it always reset settings for you via web? And what does the output of `crash` terminal command looks like? (preferably using telnet / serial instead of Debug panel)

xoseperez, 2018-10-09:
Version 1.13.3 uses a RESTFUL approach, so changing relays must be done using PUT requests. If you want to revert to the previous usage you can change it from the ADMIN tab in the Web UI or settin the `apiRestFul` setting to 0:

```
set apiRestFul 1
reload
```

xoseperez, 2018-10-09:
Nonetheless, I agree this change shouldn't have gone into a minor release.

skyynet, 2018-10-09:
I just reinstalled the 1.13.3 version and today I could save the RESTFUL API to no and use my electrodragon via ESPurna Web Intercace and from my WebApp. Dunno why it didn't work yesterday. Closed.

closed 2018-10-09 by skyynet