xoseperez/espurna #1259: Idea - Setting default power value after reset or espurna-update

mgaida11, 2018-10-09:
Hi there, 

is there  a simple (enduser-) way to set the energy-value by hand, so that the device uses this instead of 0 (zero) when an update is made (and continues to work with this then)?

I use in the mean time four supported energy counting devices and just because you guys make espurna possible. I wouldn't never ever post my data into some cloud in China or elsewhere. Instead I have managed to set up openhab. So the energy-value reported to openhab is a central value for my happiness and surveilence of my energy-usage I look up really often. When it is set to zero, my cumulative energy value makes less sense.

have a nice day, you all
Michael

xoseperez, 2018-10-09:
You can force the device to store the last energy value so it will resume counting after a reboot. This wears out the flash quite fast thou so it's not enabled by default. To do it you must define the interval between saves in the "Save every" input box in the SENSORS tab.

For instance, assuming you are reporting energy every minute (default in ESPurna). Setting this value to 1 will store the value in memory every minute but will wear out memory faster than, say, setting this to 60. Setting it to 60 will store the current energy value every hour but in the event of a reboot, you might loose up to 60 minutes of energy counting...

mgaida11, 2018-10-10:
OK, I see.

Saving the energy-value to eeprom wears it out. One saving an hour makes 24 per day, 8760 per year. What do you think, how many years will the device last?

Since the device is quite stable (I had no problems for half a year, my first device is running now), I'd like to bypass the eeprom-issue and just update the energy-value by hand when I update the firmware. So I saved the value for a test by entering the 1 (every reporting it saves the value to eeprom) as you suggested. Then I rebooted the device. But the old energy-value was gone, the status showed 0.000 kW. What am I doing wrong, please?

xoseperez, 2018-10-17:
The setting must be enabled (value > than 0) after the reboot so it will know it has to read the previous value from config.

closed 2018-12-24 by stale[bot]