xoseperez/espurna #1260: Arduino Core from git and ldscripts

mcspr, 2018-10-09:
https://github.com/esp8266/Arduino/commit/38779149d07aecefc0e23d1452d38ee01d2437f8#diff-a57fd98a29a009fc8cfd4b56cd019b53
added **local.** prefix to ldscript, so this will no longer work:
https://github.com/xoseperez/espurna/blob/2de92f87206422f764e7689203d5c8214dd4b3f0/code/eagle.flash.1m0m2s.ld#L19

using pio 'feature/stage' and trying to build anything:
```
c:/users/maxim/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open linker script file ../ld/eagle.app.v6.common.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
```

xoseperez, 2018-10-17:
Mmm... any suggestions on how to fix this in a backwards compatible way?

mcspr, 2018-10-19:
Generate ldscripts on-demand? 🌳 
https://github.com/mcspr/espurna/commit/5c9b456b959de85a42255fcd40ae8d4120e0cdf5 
https://travis-ci.org/mcspr/espurna/builds/443474512

xoseperez, 2018-11-26:
Sure, and it would not be difficult with PlatformIO. But most people use Arduino IDE so they would have to run a script manually to set up the right ldscripts depending on the core they are using...

mcspr, 2018-11-27:
Referencing these:
https://github.com/esp8266/Arduino/blob/a501d3ca3bc1eed5d0bc31b4fd5190b0a50fcb7c/platform.txt#L40 
https://github.com/esp8266/Arduino/blob/a501d3ca3bc1eed5d0bc31b4fd5190b0a50fcb7c/boards.txt#L3083-L3119
There is specific menu for flash size.

Given these files paths, they must be edited / overwritten manually by the user anyways to reference correct files. That is likely by using portable version of ide with specific version of Core installed under /hardware.
.ld scripts themselves likely can be kept in the project dir (not tested) or, if not, copied to core/tools/sdk/ld

mcspr, 2018-12-01:
Turns out there is no need to replace boards.txt / platform.txt, because IDE actually supports hardware override files.

Format is `{boards,platform}.local.txt`
Must (?) be inside the same dir that has original file, so it is either
`<ide-dir>/hardware/esp8266com/esp8266/` when using git version (advised path from esp8266/Arduino)
`<ide-data-dir>/packages/esp8266/hardware/esp8266/{2.3.0...2.5.0}/` when installed via boards manager
boards manager files go the same dir as this: https://www.arduino.cc/en/Hacking/Preferences

 Either adding separate board that will be listed in the menu, or overriding existing ones.
e.g. (need to check what keys are actually required)
```ini
espurna_1m.name=ESPurna 1M
...
espurna_1m.menu.eesz.1M=1M (1 EEPROM sector / no SPIFFS)
espurna_1m.menu.eesz.1M.build.flash_size=1M
espurna_1m.menu.eesz.1M.build.flash_size_bytes=0x100000
espurna_1m.menu.eesz.1M.build.flash_ld=espurna.flash.1m1s.ld
...
espurna_4m.menu.eesz.4M1M4S=4M (1M SPIFFS / 4 EEPROM sectors)
espurna_4m.menu.eesz.4M1M4S.build.flash_size=4M
espurna_4m.menu.eesz.4M1M4S.build.flash_size_bytes=0x400000
espurna_4m.menu.eesz.4M1M4S.build.flash_ld=espurna.flash.4m1m4s.ld
...
```
As mentioned in the previous comment, this limits the size of changes.
- find required .ld / .local.txt and place them appropriately into under the hw path
- select "ESPurna..." board and build

*ref:*
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#boardslocaltxt
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#platformlocaltxt

mcspr, 2019-02-06:
https://github.com/esp8266/Arduino/commit/951aeffa765605f30ae144a1075fac9c93fa199e 2.5.0 is here [👋](https://github.com/esp8266/Arduino/releases/tag/2.5.0)
Pending pio platform update

Also, maybe it is more appropriate to appropriate spiffs space instead of shifting rom boundary e.g. make small spiffs, like 4k (2 sectors), 12k (4 sectors)?

mcspr, 2019-06-28:
@xoseperez This is still pending as 1.13.6, but do we *reeeealy* need this? Or this can just be automagic of eeprom_rotate and some reordering always forcing a write to the real (last) sector?
I still kind of like #1559 as a general example for Arduino IDE configuration, but managing ldscript files might be confusing thing to do
(unless magic happens and we can use partitions, but it is not possible on current arduino-esp32 too :/ https://github.com/espressif/arduino-esp32/issues/2258)

PR #1559 (2019-08-05): .ld scripts refactoring
Fixes #1260 

- distribute two sets of ldscripts
- automatically pick proper ldscript via platformio python extra script
- additional flash sizes for Generic board
- float support ON/OFF menu option for IDE
- ldscript generator (sector calculation included, based on boards.txt.py script from Core)

#### Arduino IDE (only tested with 2.5.0 release):
copy proper ldscripts into the `<esp8266-package-dir>/tools/sdk/ld`
copy boards.local.txt into the `<esp8266-package-dir>`
run / restart IDE, select `ESPurna board` in the menu

#### PlatformIO
same as above, but happens auto [... cut]
closed 2019-08-05 by mcspr