xoseperez/espurna #1261: Update config changes

inverse, 2018-10-09:
I updated my devices to the latest version and so far have noticed two changes.

- Energy units got reset to J 
- Energy count got reset

I'm using my own build based on the same tag as current just with the custom.h to disable alexa + provide default configurations.

xoseperez, 2018-12-11:
The latest version added option to store energy count in flash and changed the energy unit setting name. So it's is normal that count got reset but it should have kept the units. 

Unfortunately, I cannot reproduce this now. I can see that the setting name was correctly migrated so there is no reason to think it was an omission...

closed 2018-12-11 by xoseperez

inverse, 2018-12-11:
Thanks for the update and thanks for making this awesome project!