xoseperez/espurna #1262: Conf is lost after changing mqtt topic

agevant, 2018-10-09:
An Arduino IDE 1.8.5 compiled version (striped down as per below) of Espurna 1.13.2 on Sonoff Basic  erases all previous configuration done in the web interface if  mqtt topic changes from {hostname} to one in the form of C1000/S001. The reconfiguration will work fine provided the mqtt topic stays {hostname}. More on Gitter 9/Oct/2018 11:33

//--------------------------------------------------------------------------------
// Features (values below are non-default values)
//--------------------------------------------------------------------------------

#define ALEXA_SUPPORT          0
//#define API_SUPPORT            0
//#define BROKER_SUPPORT         0
//#define BUTTON_SUPPORT         0
#define DEBUG_SERIAL_SUPPORT   0
#define DEBUG_TELNET_SUPPORT   0
//#define DEBUG_UDP_SUPPORT      1
//#define DEBUG_WEB_SUPPORT      0
#define DOMOTICZ_SUPPORT       0
#define HOMEASSISTANT_SUPPORT  0
//#define I2C_SUPPORT            1
//#define INFLUXDB_SUPPORT       1
//#define IR_SUPPORT             1
//#define LED_SUPPORT            0
//#define LLMNR_SUPPORT          1  // Only with Arduino Core 2.4.0
//#define MDNS_CLIENT_SUPPORT    1
//#define MDNS_SERVER_SUPPORT    0
//#define MQTT_SUPPORT           0
//#define NETBIOS_SUPPORT        1  // Only with Arduino Core 2.4.0
//#define NOFUSS_SUPPORT         1
//#define NTP_SUPPORT            0
//#define RFM69_SUPPORT          1
//#define RF_SUPPORT             1
#define SCHEDULER_SUPPORT      0
//#define SENSOR_SUPPORT         1
//#define SPIFFS_SUPPORT         1
//#define SSDP_SUPPORT           1
#define TELNET_SUPPORT         0
#define TERMINAL_SUPPORT       0
#define THINGSPEAK_SUPPORT     0
//#define UART_MQTT_SUPPORT      1
//#define WEB_SUPPORT            0

xoseperez, 2018-10-17:
Could it be that you are using a different memory layout from what it originally was?

agevant, 2018-10-23:
Xose, I just compiled the current 1.13.3 version. Since the pull-down menu on the Ardunio IDE 1.8.7 (for the board and memory map) differs from what it is  the current wiki page (Arduino IDE install), please verify if the selected options are ok. Upon your confirmation or not, I will recompile accordingly an go OTA to find out if this behaviour continues
[image]

agevant, 2018-10-24:
I am not sure if the previous settings are the ones that should be chosen, but I tried them and seems like it will do the trick for this particular issue. I will say, though,that the wiki page needs a pull down menu screen capture up to the current Arduino IDE version. Thank you Xose!

closed 2018-10-24 by agevant