xoseperez/espurna #1265: SETTINGS_AUTOSAVE and new delayed eeprom commit

mcspr, 2018-10-10:
When autosave is disabled it removes the commit call from the saveSettings():
https://github.com/xoseperez/espurna/blob/88a7076ad3b906f8aeee0ae6978332f0e61b4335/code/espurna/settings.ino#L367-L371

But it still will eventually save it anyways. For example, with relaySave (when switching relay)
https://github.com/xoseperez/espurna/blob/88a7076ad3b906f8aeee0ae6978332f0e61b4335/code/espurna/relay.ino#L415-L422
Or relay boot event (setSetting would already set "hostname" and "board")
https://github.com/xoseperez/espurna/blob/88a7076ad3b906f8aeee0ae6978332f0e61b4335/code/espurna/relay.ino#L549-L551

Should it be more restrictive setting, like `EEPROM_AUTOCOMMIT`? Keeping the EEPROM buffer but do not save unless explicitly told to.
Because it will be possible to set such setting, would it make sense to keep some state in RTC? Like relay state between reboots. It should keep things around unless device completely powers off or gone into deep sleep

xoseperez, 2018-10-17:
AFAIK, RTC memory does not survive across reboots: "The stored data can be retained between deep sleep cycles. However, the data might be lost after power cycling the chip."

The main reason not to commit the data right away is to save writing cycles. Using a force commit to write the rest of the buffer is OK.

mcspr, 2018-10-19:
Oh, I confused RTC timer with RTC mem itself. Citing NONOS_SDK pdf:
> System time will return to zero because of `system_restart` (note: === `Esp.restart()`), but RTC still goes on.
If an external hardware reset the chip via EXT_RST or CHIP_EN (such as timed
wakeup from Deep-sleep), the RTC timer will be reset.
• Reset by pin EXT_RST: RTC memory won’t change; RTC timer starts to zero.
• Watchdog reset: RTC memory won’t change; RTC timer won’t change.
• system_restart: RTC memory won’t change; RTC timer won’t change.
• Power-on: RTC memory contains a random value; RTC timer starts from zero.
• Reset by pin CHIP_EN: RTC memory contains a random value; RTC timer
starts from zero.

And about data what I meant is having EEPROM in Read-Only state forever. Essentially just treat that flash segment as some initial state to work with.

mcspr, 2018-11-05:
I did a preliminary test for RTC memory storage and it works OK. Some things to clean-up before PR though, I'll leave technical implementation comments there. And it also touches system stability logic, as currently it relies on loop actually finishing to commit to go through.

Important thing to note about RTC memory - arduino bootloader uses part of user region to store data when doing OTA:
https://github.com/esp8266/Arduino/blob/74819a763bfb6e9890a57411dcea4aba221a778d/bootloaders/eboot/eboot_command.h#L25-L30
https://github.com/esp8266/Arduino/blob/74819a763bfb6e9890a57411dcea4aba221a778d/bootloaders/eboot/eboot_command.c#L48-L58
https://github.com/esp8266/Arduino/blob/e876170cf4d31528420d201ed059de35a3e75b78/cores/esp8266/Updater.cpp#L199
So actual available storage is not 512 bytes, but 384. Starting from the 128th byte.

xoseperez, 2018-12-09:
INteresting, have you done any progress on this? 384bytes would be enough to hold data that are responsible for most commits (relay status, colors,...)

mcspr, 2018-12-10:
Yep, only haven't decided how to actually arrange access to it. Three tested options:
- pick numeric defines / typed enum ranging 31...127 (i.e. block address) to store the data as 32bit uints + allow ranged reads (32bit * N times). on module basis convert back to something usable
- directly use rtc mem just pointing to `0x60001201 + 128` as struct of the same uint32 blocks. slightly nicer. *may* break alignment and I haven't tested if this works using something like union inside / does this require aligned(4) hint
- global struct for full 384 bytes. just read it whole and keep in memory all the time, periodically syncing.

mcspr, 2018-12-10:
Aand I'll answer my own question - it does work ok directly using struct (and using it as volatile seems enough so compiler doesn't do something strange)
I'll prepare this variant as PR with relay state + syscheck

mcspr, 2018-12-12:
BTW, related to #1420 - I did remember that esp-idf does things slightly differently with rtc.
Referencing RTOS SDK for esp8266 (espressif converting it to be more like idf):

`rtc_seg` section:
https://github.com/espressif/ESP8266_RTOS_SDK/blob/c10e6d39067595e82bc92d3852df7b844cdf788c/components/esp8266/ld/esp8266.ld#L33

`.rtc.data` section definition:
https://github.com/espressif/ESP8266_RTOS_SDK/blob/c10e6d39067595e82bc92d3852df7b844cdf788c/components/esp8266/ld/esp8266.common.ld#L67-L74

Main magic, `RTC_DATA_ATTR` attribute macro:
https://github.com/espressif/ESP8266_RTOS_SDK/blob/1e294d110262625d6e8df0b4c0ceb5fe9f2f6b58/components/esp8266/include/esp_attr.h#L44

This makes generic global variable be placed in the RTC section by the compiler itself. Alignment + length checks included (via section definition)