xoseperez/espurna #1266: PLUGIN: procedure does not work with arduino ide

lblabr, 2018-10-10:
it seem that 

../../lib/PLUGIN/plugin.h

is not found by arduino ide....

sketch\config/custom.h:120:42: fatal error: ../../lib/mcp4725/mcp4725.h: No such file or directory

    #include "../../lib/mcp4725/mcp4725.h"

                                          ^

compilation terminated.

xoseperez, 2018-10-17:
The plugin module is a WIP. Actually, it shouldn't be in the master branch...

lblabr, 2018-10-18:
but the description told me to place it in lib that's not working if you use arduino ide

"Copy plugin1.h files into lib/plugin1 folder (create plugin1 folder in existing lib)"
we are talking about <espurna base>/code/lib ??

lblabr, 2019-01-13:
i would like to do another plugin, how do i contribute my experiences with the plugin concept ?
i'm new to git, in the past i just downloaded, how can i participate ? could you point me in the correct direction ?

thanks in advance!