xoseperez/espurna #1267: PLUGIN: wsOnAfterParseRegister is not available

lblabr, 2018-10-11:
C:\Users\Lars Bretschneider\lblabr@owncloud\Projekte\Arduino\github.espurna\code\espurna\mcp4725.ino: In function 'void _pluginSetupWS()':

mcp4725:158:57: error: 'wsOnAfterParseRegister' was not declared in this scope

           wsOnAfterParseRegister(_pluginWebSocketOnParse);

                                                         ^

exit status 1
'wsOnAfterParseRegister' was not declared in this scope

mcspr, 2018-10-11:
Make sure to reference current code: 7d09f59952597d07c042e41867c1928f0a41520d
If that one happens to be out of scope - add its declaration to 'espurna/config/prototypes.h' e.g ```void espurnaReload();```

lblabr, 2018-10-11:
i have taken the plugin.zip....

i'm not that expert in programming.

just make clear for me, wsOnAfterParseRegister is more or less removed from framework, so remove it from my plugin template ?? at the moment i´m fine with that

br lars

mcspr, 2018-10-11:
plugin template from wiki is kinda complicated to keep it up-to-date. i'd reference any other integration within repo instead - at least it is expected to be working.

if you look at the commit changes - wsOnAfterParseRegister is renamed to espurnaRegisterReload. and it no longer depends on websocket to work.

lblabr, 2018-10-11:
for me and maybe others it would be great to have a template to integrate features....
an idea could maybe to add an folder for the template in the repro, so that other can contribute to it and to keep up to date .....

i got my i2c DAC working with it and i´m very happy.... i just need to integrate in web ui i that very difficult for me to understand hwo to do that ...

thank you very much for responding all the time! maybe someone can review my code and make it availabe also for others....

mcspr, 2018-10-12:
Great! Check out recent blynk pr on how it added panel + config: https://github.com/xoseperez/espurna/pull/1230/files#diff-35d45b13e9460cb72e90b29e1c6e56a0
Note that some inputs' "name" attributes are directly mapped to configuration keys (see `config` in terminal or try do config backup)

lblabr, 2018-10-12:
can you explain a litte bit more ?

thanks in advance 

lars

lblabr, 2018-10-12:
great would be, if i could add an ui element with a plugin...

closed 2019-02-14 by stale[bot]