xoseperez/espurna #1269: Enhancement request: one for the hardware folks

JohnOH, 2018-10-11:
Might it be possible to connect a [current transformer](https://en.wikipedia.org/wiki/Current_transformer) (CT) to a Sonoff POW2. I would then hope that the frequency of the AC signal could be derived from the signal generated by the CT. I think this would complete the POW2 as a piece of hardware/

Valcob, 2018-10-11:
@JohnOH please don't play with mains it can hurt/kill you but if you know what you are doing then I can suggest you using a optocoupler like [Vishay's SFH620](http://www.vishay.com/docs/83675/sfh620a.pdf) and a simple schematic to count the so cold zero crossings with a simple counter sensor that will give you the missing functionality. the schematics is simple  you can read more here [schematic](https://electronics.stackexchange.com/questions/33042/how-do-i-select-the-accompanying-components-for-an-optocoupler)  
You can buy optocouplers here:
[in DIP 4 case](http://s.click.aliexpress.com/e/z30gpx6)
or
[in SMD case](http://s.click.aliexpress.com/e/ytxtB3K)

PS: please don't use current transformer (I do believe you are talking about a simple voltage transformer though) ;) have fun

JohnOH, 2018-10-11:
I was hoping that the CT would keep the AC at more than arms length. Perhaps you mean the unusual ground plane on the POW2.