xoseperez/espurna #1271: Mainline Sonoff SC support?

Cabalist, 2018-10-11:
I was curious if the Sonoff SC will be supported by the main Espurna repo?   I am using an older version from https://github.com/xoseperez/sonoffsc and miss some of the nice features of the updated codebase.

xoseperez, 2018-10-17:
It is a work in progress. Check the SonoffSC branch.

Cabalist, 2018-10-17:
Thanks!  You are doing amazing work!

peterhoeg, 2018-11-22:
I'm super keen to try this out!

peterhoeg, 2021-03-08:
@mcspr, are you by chance doing any work on getting this integrated?

mcspr, 2021-03-09:
No. I also did a double-check, we don't have anything related to AT commands handling atm.
Did you mean to make the port?

peterhoeg, 2021-03-09:
I do not realistically have time to work on that. I just happen to have one of these devices sitting in a drawer and would like to start using it.

mcspr, 2021-03-10:
Most of the code is already there. SC-specific support consists of comms.ino + serial link helper functions that talk through the AT commands, everything else is *supposed* to be a sensor i.e. commsLoop() -> BaseSensor::tick(), read-out functions -> BaseSensor::value(...) and etc.

I could look at it some more, but no promises it will be soon

peterhoeg, 2021-04-21:
That would be awesome!