xoseperez/espurna #1273: HA Config gives errors for rgb values

markcarline, 2018-10-13:
Using the latest firmware - espurna-1.13.3-aithinker-ai-light.bin

My ESP light seems to work fine and is defined in HA (copied from bulb suggestion) as:

```
    - platform: mqtt
      name: "LEDBULB02"
      state_topic: LEDBULB02/relay/0
      command_topic: LEDBULB02/relay/0/set
      payload_on: 1
      payload_off: 0
      availability_topic: LEDBULB02/status
      payload_available: 1
      payload_not_available: 0
      brightness_state_topic: LEDBULB02/brightness
      brightness_command_topic: LEDBULB02/brightness/set
      rgb_state_topic: LEDBULB02/rgb
      rgb_command_topic: LEDBULB02/rgb/set
      color_temp_command_topic: LEDBULB02/mired/set
      white_value_state_topic: LEDBULB02/channel/3
      white_value_command_topic: LEDBULB02/channel/3/set

```

Problem is that i seem to get errors being reported in the HA log files as per:

```
Log Details (ERROR)
Sat Oct 13 2018 12:41:08 GMT+0100 (British Summer Time)

Error doing job: Exception in callback MQTT._mqtt_handle_message(<paho.mqtt.cl...at 0x61d87030>)
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/mqtt/__init__.py", line 749, in _mqtt_handle_message
    subscription.callback, msg.topic, payload, msg.qos)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/core.py", line 327, in async_run_job
    target(*args)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/light/mqtt.py", line 313, in rgb_received
    rgb = [int(val) for val in payload.split(',')]
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/light/mqtt.py", line 313, in <listcomp>
    rgb = [int(val) for val in payload.split(',')]
ValueError: invalid literal for int() with base 10: '#FF7F00'

```

I managed to snif the MQTT traffic and get this:

```
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/app', ... (7 bytes))
ESPURNA
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/version', ... (6 bytes))
1.13.3
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/board', ... (18 bytes))
AITHINKER_AI_LIGHT
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/host', ... (9 bytes))
LEDBULB02
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/ip', ... (13 bytes))
192.168.1.142
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/mac', ... (17 bytes))
A0:20:A6:2B:9E:F4
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/rssi', ... (3 bytes))
-72
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/uptime', ... (3 bytes))
305
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/datetime', ... (19 bytes))
2018-10-13 13:41:08
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/freeheap', ... (5 bytes))
22904
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/relay/0', ... (1 bytes))
1
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/rgb', ... (7 bytes))
#FF7F00
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/hsv', ... (9 bytes))
30,100,98
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/mired', ... (3 bytes))
326
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/channel/0', ... (3 bytes))
255
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/channel/1', ... (3 bytes))
127
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/channel/2', ... (1 bytes))
0
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/channel/3', ... (1 bytes))
0
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/brightness', ... (3 bytes))
250
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/vcc', ... (4 bytes))
3208
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/status', ... (1 bytes))
1
Client mosqsub/813-raspberrypi received PUBLISH (d0, q0, r0, m0, 'LEDBULB02/loadavg', ... (1 bytes))
1

```

Is this some kind of issue with the formatting in HA or the format of what the bulb sends to MQTT for the RGB value ?

Thanks

Mark

markcarline, 2018-10-13:
UPDATE:  I posted a note over on the HA forum to see if anyone over there knows.

https://community.home-assistant.io/t/espurna-config-giving-rgb-errors/72844

Once we know what it should be I think the firmware needs to be updated so it suggests the correct config.

Thanks!

mcspr, 2018-10-17:
Yep, it does not support CSS syntax: https://www.home-assistant.io/components/light.mqtt/#rgb_state_topic
Which raises the question - should it just maybe warn about "useCSS" setting when showing config? Overriding might be too much, hass module can be enabled in espurna .bin but #css syntax still can have other consumers like node-red.

xoseperez, 2018-11-19:
Maybe the RGB color should be published in both syntaxes under different topics?

closed 2019-02-11 by stale[bot]