xoseperez/espurna #1276: MQTT Device always unavailable

wardaddyalpha, 2018-10-15:
Hi,

I enabled MQTT and HASS in ESPurna UI and setup my Home Assistant accordingly for the MQTT broker and yaml files, however the Home Assistant always keeps reporting my devices as 'Unavailable' and last seen a few seconds ago. The Debug log in ESPurna report that it successfuly connected to the MQTT broker and then says disconnected. I am not able to control the device from Home Assistant as its never online. Retain flag is enabled and alive time is 12hrs. Tried with the defaults too.

`[765084] [MQTT] Connecting to broker at http://192.168.2.111:1883
[765087] [MQTT] Connecting as user owner
[765089] [MQTT] Client ID: ESPURNA-3AA300
[765090] [MQTT] QoS: 0
[765091] [MQTT] Retain flag: 1
[765093] [MQTT] Keepalive time: 43200s
[765096] [MQTT] Will topic: smartroomlight/status
[765134] [MQTT] TCP Disconnected
[765136] [MQTT] Disconnected!`

This is the yaml config:

`mqtt:
  broker: core-mosquitto
  username: wardaddy
  password: wardaddy86
  discovery: true
  discovery_prefix: homeassistant
  
switch:
  - name: smartroomlight
    platform: mqtt
    state_topic: 'smartroomlight/relay/0'
    command_topic: 'smartroomlight/relay/0/set'
    payload_on: 1
    payload_off: 0
    optimistic: false
    qos: 0
    retain: true
    availability_topic: 'smartroomlight/status'
    payload_available: 1
    payload_not_available: 0`

I've tried adding and removing retain,optimistic,payload_availaibie flags but no difference.

Another issue I see in Home Assistant log is that it is not able to fetch the setup info

`Error fetching description at http://192.168.2.52:52002/setup.xml
11:28 PM ssdp.py (WARNING)
Error fetching description at http://192.168.2.52:52001/setup.xml
11:28 PM ssdp.py (WARNING)
Error fetching description at http://192.168.2.52:52000/setup.xml
11:28 PM ssdp.py (WARNING)`

mcspr, 2018-10-16:
> [765084] [MQTT] Connecting to broker at http://192.168.2.111:1883
You might want to remove http://

wardaddyalpha, 2018-10-19:
Oh man! As simple as that! saved me a lot of trouble my friend! Thank you very much!

closed 2018-11-26 by xoseperez