xoseperez/espurna #1277: A few questions about use BMP280 and CT transformer in Espurna

xury77, 2018-10-15:
Hi.
I had read viki and successfully configured compile flags to use i2c . I have BMP280 sensor and ADS1115 ADC connected to wemos d1 mini. Both of devices are working but I don't know how to calibrate pressure sensor to display an absolute value. I cannot find settings to set sea level so sensor showing value 24hPa lover. My sensor is about 185 meter above sea level. Is a easy solution in espurna for this?
Second question: I want to measure current value using SCT 013 30 connected to ADS1115.
I manage to enable it in sensors.h but I need three values only. Were I can switch of 4th channel?
And how to calibrate it. There is some factor settings?

xoseperez, 2018-10-17:
There is no way to set an offset for the pressure (or the height above see level) for the BME280 but it's a good enhancement.
Yes, the comment is wrong. It should be 30A/1V. ATM that is the only way to calibrate the sensor and there is no way to set different ratios for the different channels. Even thou the code is already there, the configuration does not allow it:

https://github.com/xoseperez/espurna/blob/88a7076ad3b906f8aeee0ae6978332f0e61b4335/code/espurna/sensor.ino#L448-L451

xury77, 2018-10-20:
Thank you for your reply. But I still don't know how to calibrate it.  I have three claps connected to ads1115.
I want to measure three phase current using SCT-013-30 clamps. The ratio is 30 as default setting. Clamps are witch burden resistor already and giving real value 1V on 30A. But the current values in Espurna is very too high ( about 120A, but should be about 100mA). When I stop the current Espurna showing about 0A (its OK).

xoseperez, 2018-11-26:
Sorry for the late reply. 
I can't see a reason for this... let me try it.

mcspr, 2019-10-22:
> https://github.com/xoseperez/espurna/blob/88a7076ad3b906f8aeee0ae6978332f0e61b4335/code/espurna/sensor.ino#L448-L451

So, this should be overridable EMON1_CURRENT_RATIO, EMON2_CURRENT_RATIO, etc.
At the same time there is HLW8012_CURRENT_RATIO, which should probably be EMON1_CURRENT_RATIO alias?
Looking at v2 branch, it has drafted out `curRatio` and `curRatio0` as a separate keys though.

There is another question of how to approach this in generating numerical ID for ratio keys, should energy monitor sensor be a single one? Or should it use magnitude's global indexing value to properly offset different sensors?

I already marked reset and calibration as something TO-DO, not really liking config keys as triggers for this, so fixing emon indexes would probably go with that issue too