xoseperez/espurna #1282: No issue with ESPURNA but with cheap DS18B20 sensors

AlbertWeterings, 2018-10-17:
As most of you know on aliexpress many cheap DS18B20 sensors are sold and a lot of them are (counterfeit) fake. I have got myself a lot of them. Having that said I will sum up some problems you might run into using this sensors on ESPURNA.

1. A Sonoff TH10 has internaly two 10K resistors in parallel from VCC 3v3 to IO pin 14 resulting in a 5K resistance. So if you use a genuine or fake sensor no 4K7 resistor is needed.
2. If you connect a fake sensor for example to a Sonoff TH10 and you use a resistor of 4k7 from VCC to DO you will see the temperature rising over time this is due to the sensor heating up its self rapidly. The 4k7 resistor makes it worse.
3. Most Fake sensors also heat up with the Sonoff's internal resistance of 5K
4. Replacing the internal two 10K resistors with one 50K resistor stops the rapid heating up. but measurement is still in accurate.
5. If you have your fake DS18B20 mounted on a surface or floating in fluid it will not heat up rapidly anymore but still is giving inaccurate measurements as it is heating its self internally.

I was unable to find a setup in witch the fake sensors where measuring accurate so my advise if you ever have issues with these type of sensors compare it to a sensor that is known genuine.

xoseperez, 2018-11-26:
I think this could be a page in the wiki. Maybe we could open a new section about sensor-specific issues and troubleshooting. Some might be generic issues or recommendations when using them, others might be ESPurna or ESP8266 specific.

AlbertWeterings, 2018-11-30:
I think that is a good idea.

xoseperez, 2018-12-08:
Added to https://github.com/xoseperez/espurna/wiki/Sensors#dallas-1-wire-sensors for the moment.

closed 2018-12-08 by xoseperez