xoseperez/espurna #1283: Support for Teckin SP22 V1.4?

steve8x8, 2018-10-17:
There's a new make sold under the Teckin brand, which lacks the small "T" logo. It's incompatible with the old Teckin devices (of the Blitzwolf kind).

Inside, it's got a green silkscreen, and a TYWE2S chip mounted in one of the corners.
To connect wires to the ESP, it may be necessary to unsolder the mains wires.
Tasmota seems to have support for this variant: https://github.com/arendst/Sonoff-Tasmota/issues/3950

Can we have Espurna support for this device?

ColinShorts, 2018-10-17:
I'm not sure about the LED configuration - particularly the number/function of LED's, but this should get you started:

code/espurna/config/arduino.h:
`//#define TECKIN_SP22v1.4`

code/espurna/config/hardware.h:
```

// -----------------------------------------------------------------------------
// Teckin SP22 V1.4
// https://www.amazon.de/gp/product/B07D5V139R
// -----------------------------------------------------------------------------

#elif defined(TECKIN_SP22v1.4)
    // Info
    #define MANUFACTURER                "TECKIN"
    #define DEVICE                      "SP22v1.4"

    // Buttons
    #define BUTTON1_PIN                 1
    #define BUTTON1_MODE                BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY               1

    // Relays
    #define RELAY1_PIN                  14
    #define RELAY1_TYPE                 RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN                    0
    #define LED1_PIN_INVERSE            0
    #define LED2_PIN                    2
    #define LED2_PIN_INVERSE            0
    #define LED3_PIN                    3
    #define LED3_PIN_INVERSE            1
    #define LED3_RELAY                  1

    // HJL01 / BL0937
    #ifndef HLW8012_SUPPORT
    #define HLW8012_SUPPORT             1
    #endif
    #define HLW8012_SEL_PIN             12
    #define HLW8012_CF1_PIN             5
    #define HLW8012_CF_PIN              4

    #define HLW8012_SEL_CURRENT         LOW
    #define HLW8012_CURRENT_RATIO       25740
    #define HLW8012_VOLTAGE_RATIO       313400
    #define HLW8012_POWER_RATIO         3414290
    #define HLW8012_INTERRUPT_ON FALLING

```

code/espurna/migrate.ino:
```

        #elif defined(TECKIN_SP22v1.4)

            setSetting("board", 92);
            setSetting("ledGPIO", 0, 0);
            setSetting("ledLogic", 0, 1);
            setSetting("ledGPIO", 1, 2);
            setSetting("ledLogic", 1, 0);
            setSetting("ledGPIO", 2, 3);
            setSetting("ledLogic", 2, 1);
            setSetting("ledRelay", 2, 0);
            setSetting("btnGPIO", 0, 1);
            setSetting("btnRelay", 0, 0);
            setSetting("relayGPIO", 0, 14);
            setSetting("relayType", 0, RELAY_TYPE_NORMAL);
            setSetting("selGPIO", 12);
            setSetting("cf1GPIO", 5);
            setSetting("cfGPIO", 4);
            setSetting("pwrRatioC", 25740);
            setSetting("pwrRatioV", 313400);
            setSetting("pwrRatioP", 3414290);
            setSetting("hlwSelC", LOW);
            setSetting("hlwIntM", FALLING);

```

code/platformio.ini:

```
[env:teckin_sp22v1.4]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_1m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_1m0m} -DTECKIN_SP22v1.4
upload_speed = ${common.upload_speed}
monitor_speed = ${common.monitor_speed}
extra_scripts = ${common.extra_scripts}

[env:teckin_sp22v1.4-ota]
platform = ${common.platform}
framework = ${common.framework}
board = ${common.board_1m}
board_build.flash_mode = ${common.flash_mode}
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags_1m0m} -DTECKIN_SP22v1.4
upload_speed = ${common.upload_speed}
monitor_speed = ${common.monitor_speed}
upload_port = ${common.upload_port}
upload_flags = ${common.upload_flags}
extra_scripts = ${common.extra_scripts}

```

ColinShorts, 2018-11-05:
@steve8x8  did you get around to trying this?

steve8x8, 2018-11-10:
Hi, just returned from some overdue vacation, to charge the inner battery :)
TBH, I didn't get very far yet, and I'm still investigating what might be wrong. No point in changing codes yet :/
I have opened the device (without major damage to it but my hands), clipped the power wires, and soldered the 5 wires required for the serial console. After connecting them to a CH340G properly set to 3.3V, I don't get any output when powering up the board.
Maybe the CH340 is faulty, maybe something else is fishy. `esptool chip_id` times out, `screen /dev/ttyUSB0` doesn't show anything. Same with, and without IO0 connected to GND.
Any suggestions what else to check? Should I fetch an oscilloscope to verify that the UART is sending data during startup?

xoseperez, 2018-11-26:
I have it working with a new Teckin SP22 v1.6. I hope it's compatible with the v1.4. I'll commit the changes this week.

Wikibear, 2018-12-04:
I'm with you. Opening of housing was very tricky and i see a TYWE2S chip. Found Datasheet:
https://fccid.io/2ANDL-TYWE2S/User-Manual/Users-Manual-3596121.pdf
There you will find all connections we need. Maybe reset first via power button hold for 5 sec to get it on. At this stage i doesn't connect anything. Will do it within the next days. I don't have a clue to get this chip into flash mode. Would be cool if someone can share some nice stuff here.

xoseperez, 2018-12-07:
Yes it works, relay, LEDs, button and energy monitoring. It's already available in the `dev` branch.

closed 2018-12-07 by xoseperez

mgaida11, 2019-01-05:
Wikibaer, did you find a solution to your flashing problem? I have the same at the moment and absolutely no idea what to do.

Wikibear, 2019-01-05:
You must flash this like each other ESP. I've tested it yesterday and today. If you have opened the plug then the contacts show to the middle are the right ones. I use external supply 3,3V and use wires. TX RX is not contacted. Check my PDF. On TYWE2S there is IO0. You must shurtcut this contact to ground.

https://i.imgur.com/mYUKzPL.png

You must solder TX RX directly to the chip, then there is no need to remove short wires. It's a littlebit tricky but it works. There is an AMS1117 3,3V stabelizer and gnd is every where. You can measure it.

If TX RX is solderd and other one then you can read out normaly via COM and 74880 the ESP Data. If you can see this, then you have all connected right. Now put in flash mode: Power off IO0 to GND and power on. You can't solder IO0 i do it with a wire and a outlet strip and a switch button. Now you can see only short Text via COM. The boot loader show now 1,7. This is the flash mode. Now you can flash your device.

mgaida11, 2019-01-05:
@Wikibaer: The scripts, the guy is mentioning, would solve my problem. And the thinks he found out make me fear about the integrity of almose any home. It lets me stick to the Espurna-community! Thanks for the link!

Wikibear, 2019-01-05:
Yep, that's why don't use stock firmware from chines...