xoseperez/espurna #1284: sonoff T1 R2 EU instability

FrColin, 2018-10-17:
I have just flash with the 1.13.3
and the sonoff bip and crash

[005507] [NTP] UTC Time  : 2018-10-17 20:56:01
[005507] [NTP] Local Time: 2018-10-17 22:56:01
[023972] [BUTTON] Button #0 event 1
[023973] [MQTT] Sending prise-E/button/0 => 1 (PID 1)
[023974] [RELAY] #0 scheduled ON in 0 ms
[023985] [RELAY] #0 set to ON
[023987] [MQTT] Sending prise-E/relay/0 => 1 (PID 1)
[024988] [RELAY] Setting relay mask: 1
[034059] [BUTTON] Button #0 event 5
[034060] [MQTT] Sending prise-E/button/0 => 5 (PID 1)

Exception (3):
epc1=0x4010011d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40035550 depc=0x00000000

ctx: cont 
sp: 3fff1930 end: 3fff1c30 offset: 01a0

>>>stack>>>
3fff1ad0:  00000000 3fff1b60 402319b9 3ffef150  
3fff1ae0:  3fff1d24 000005fd 000005fd 4010020c  
3fff1af0:  3fff0870 00000000 3fff25fc 4010068c  
3fff1b00:  3fff1b40 00000000 3fff0870 4010740c  
3fff1b10:  402310d2 3fff25fc 3fff25fc 4023363f  
3fff1b20:  40231f06 3fff25fc 3fff25fc 00000000  
3fff1b30:  40213eed 3fff0950 3fff0c10 00000000  
3fff1b40:  3fff0700 3fff1b90 40227a40 3fff0c10  
3fff1b50:  3fff1d24 0000041b 00000064 40227ab8  
3fff1b60:  00000000 00000000 3fff1bb0 40227af8  
3fff1b70:  3ffeeed0 3fff6114 3fff5fbc 40214ecf  
3fff1b80:  3fffdad0 3fff07b0 0000000b 402214b4  
3fff1b90:  40231f2f 00a32f28 00000000 40227fe4  
3fff1ba0:  3fff25fc 402140b0 3fff02e4 40203e8e  
3fff1bb0:  40104ac0 00000000 3fff0950 40221c7d  
3fff1bc0:  3ffeeed0 3fff1c80 3fff0c10 3fff0c04  
3fff1bd0:  3fffdad0 3fff0828 3fff0950 40222079  
3fff1be0:  402017e6 0000000a 00000003 4022215c  
3fff1bf0:  3fffdad0 3fff0828 3fff0b48 4020e75e  
3fff1c00:  00000000 3fff0828 00000003 402038fc  
3fff1c10:  3fffdad0 00000000 3fff0bfd 40227a8c  
3fff1c20:  feefeffe feefeffe 3fff0c10 401006fc  
<<<stack<<<

ways, 2018-10-22:
I've flashed two recently, none of which works reliably. Seems to me that buttons are being "pressed". Including the reset button.

I have three laying around waiting installation, so I can test anything on them if needed.

`
                            [000707]
                                      [000708] [MAIN] EEPROM sectors: 251, 250
                                                                              [000711] [MAIN] EEPROM current: 251
                [000714]
                         [000715] [MAIN] EEPROM:  4096 bytes initially |   350 bytes used ( 8%) |  3746 bytes free (91%)
                       [000724] [MAIN] Heap  : 37032 bytes initially |  5256 bytes used (14%) | 31776 bytes free (85%)
                     [000732] [MAIN] Stack :  4096 bytes initially |   928 bytes used (22%) |  3168 bytes free (77%)
                   [000740]
                            [000741] [MAIN] Boot version: 31
                                                            [000744] [MAIN] Boot mode: 1
                                                                                        [000747] [MAIN] Last reset reason: External System
                                         [000751] [MAIN] Last reset info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000  
  [000764]
           [000766] [MAIN] Board: ITEAD_SONOFF_T1_2CH
                                                     [000769] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB
                                 [000784] [MAIN] WebUI image: SMALL
                                                                   [000787]
                                                                            [000998] [MAIN] Firmware MD5: ca8714a0fe77185bbb06f28fc265d098
                                         [000999] [MAIN] Power: 4094 mV
                                                                       [000999] [MAIN] Power saving delay value: 10 ms
                     [001000]

                              ---8<-------

                                          [001009] [TELNET] Listening on port 23
                                                                                [001010] [WEBSERVER] Webserver running on port 80
                                [001013] [RELAY] Retrieving mask: 2
                                                                   [001014] [RELAY] Relay #0 boot mode 0
       [001016] [RELAY] Relay #1 boot mode 0
                                            [001019] [RELAY] #0 set to OFF
                                                                          [001022] [RELAY] #1 set to OFF
       [001026] [RELAY] Number of relays: 2
                                           [001028] [BUTTON] Number of buttons: 2
                                                                                 [001032] [LED] Number of leds: 1
                [001034] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED 
                                                                                [001046] [THINGSPEAK] Async ENABLED, SSL DISABLED
                                [001058] [MAIN] Uptime: 1 seconds
                                                                 [001058] [MAIN] Heap  : 37032 bytes initially | 11064 bytes used (29%) | 25968 bytes free (70%)
                                                               [001059] [MAIN] Power: 4094 mV
                                                                                             [001081] [WIFI] Creating access point
                                 [001271] [WIFI] Captive portal enabled
                                                                       [001273] [WIFI] -------------------------------------- MODE AP
                                    [001273] [WIFI] SSID  ESPURNA-AD3D90
                                                                        [001273] [WIFI] PASS  fibonacci
      [001275] [WIFI] IP    192.168.4.1
                                       [001278] [WIFI] MAC   62:01:94:AD:3D:90
                                                                              [001282] [WIFI] ----------------------------------------------
                                           [001288] [MDNS] OK
                                                             [002025] [RELAY] Setting relay mask: 0
  [007968] [BUTTON] Button #1 event 1
                                     [007970] [RELAY] #1 scheduled ON in 0 ms
                                                                             [007980] [RELAY] #1 set to ON
         [008470] [BUTTON] Button #1 event 2
                                            [008981] [RELAY] Setting relay mask: 2
                                                                                  [046189] [BUTTON] Button #0 event 1
                                                                                                                     [046190] [RELAY] #0 scheduled ON in 0 ms
                                                                                                                                                             [046201] [RELAY] #0 set to ON
                             [047201] [RELAY] Setting relay mask: 3
                                                                   [056448] [BUTTON] Button #0 event 5

[... cut ...]
            ---8<-------

                        [000635] [MAIN] ESPURNA 1.13.3 (e2641b9)
                                                                [000636] [MAIN] xose.perez@gmail.com
                                                                                                    [000636] [MAIN] http://tinkerman.cat

                                                                                                                                        [000636] [MAIN] CPU chip ID: 0xAD3D90
`

ways, 2018-10-25:
Could it be a reappearance of #662, floating buttons?

ways, 2018-10-30:
I've tried the three I have to spare with an older firmware, espurna-1.12.1-itead-sonoff-t1-2ch.bin. Same problem. Can it be a bad ground? Too low voltage from usb?

mcspr, 2018-10-31:
@ways Did you test when connected to mains? Without serial connection ofc, just normal operation.

FrColin, 2018-11-02:
I have restested mine without change or reflashing , it is working normaly , 
surely it's a power supply trouble.

closed 2018-11-02 by FrColin

ways, 2018-11-04:
I guess you're correct. It seems to work for me too. I had a PIR sensor connected to one of them, that caused to much draw and instability. The others seems to work on both 12.0.1 and 1.13.3.