xoseperez/espurna #1285: [Sonoff Basic] Unable to access web page - rest is ok

ccapublic, 2018-10-18:
Hi,

I am in a situation where some of my devices are connected to my wifi network, can be properly controlled (On/Off/Toggle from MQTT), but do not seem to serve web page any more. When accessing them from any web browser, I get prompted for credentials (and I am 100% I enter correct value), then browser tab displays "Espurna 0.0.0" and browser seem to endlessly wait (rolling circle) but displays nothing but blank page. The phenomenon sometimes disappears but most of times it fails. 

One important thing is that they are the most distant from my access point (show less than -70dbm). But If this was the root cause, why does MQTT work ?

Any advice appreciated (I'm running last stable release)

Thanks 

Cyril.

Valcob, 2018-10-18:
You can check the console log and check what happens to websocket in your browser tap F12, do a page refresh and check the output in console and on network tab

ccapublic, 2018-10-23:
Hi Valcob,

In the meantime I installed a Wifi repeater, raising RSSI from below -70 to roughly -50 and the problem did not show up again since. I guess mqtt packets are smaller and more immune to poor connections than eb page loading... I will give a try to your suggestion the next time the problem occurs. Thanks again !

spenjt, 2018-11-15:
Hi, I also have a Sonoff Basic and now can't access the web interface after an update via the web interface.
The sonoff seems to be working fine. It is connected via MQTT and is working as expected.
Is there anything I can do to get the web interface back?

ccapublic, 2018-11-15:
Hi,

Just to let you know that I ended up adding a WiFi Extender to ensure RSSI was always around -60 or better and since then the problem never occurred again. I am also under the impression that latency is lower and lags/stalls are now a thing of the past.

Note that at the time it failed (so before adding the extender), the web ui of a failing device  was not accessible from the outside and through my VPN.

PS :I did not try to analyze browser logs before adding the extender.  So I have no "before/after" to share.

spenjt, 2018-11-16:
I have tried the Safe Mode. (5 on and off cycles). I can log into the Espurna Wifi, but still no Web UI under 191.168.4.1

Is my only option to factory reset and start again?

Valcob, 2018-11-16:
@spenjt, well that is interesting, could you please connect the device to serial and share the log

spenjt, 2018-11-16:
Sorry, I have no idea what you mean. I have very limited knowledge.

The Sonoff is not easily accessible, which is why I haven't yet taken it out and tried to start again.

mcspr, 2018-11-16:
When you device is in AP mode, you can also connect to it via `telnet`. It is the same as serial connection. 
Just open your platform's terminal app and run `telnet 192.168.4.1`

And if you happen to use `espurna-core` - it does not have any WebUI builtin. Just update via OTA to the proper version.

spenjt, 2018-11-16:
OK, so I have managed to telnet in (This is all new to me). I have used PuTTy.

What's the best thing to try, another ota update? (as this is what seemed to break it).

This is what I used: espurna-1.13.3-itead-sonoff-th.bin for the sonoff basic. I had updated another sonoff with no problems.

The telnet info says to use the command "ota URL". What is the URL?

Valcob, 2018-11-16:
@spenjt you can type info and share with us the output. The url is the path where the bin file is like https://github.com/xoseperez/espurna/releases/download/1.13.3/espurna-1.13.3-itead-sonoff-basic.bin for e.g

spenjt, 2018-11-16:
Thank you for your help. But as it won't have access to the internet in AP mode it can't download from anywhere. I tried using a networth path to my laptop, but couldn't get it to work.

I have given up and replaced it with another switch now.

spenjt, 2018-11-18:
I reflashed the sonoff via the laptop and it seems to be working OK now.