xoseperez/espurna #1286: Wemos D1 - Button failure

moddsson, 2018-10-18:
Hi,

I have an interesting problem.  I have configured the Wemos D1 for 8 buttons and after restart all buttons respond correctly but only once.  After that only the button on D3 continues responding.  I wonder if there is something broken in the button.ino file and only the lowest button continues working?  Possibly a debouncing routine that never returns?

This is the core of my hardware.h file:

 // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1

	// Buttons
	 #define BUTTON4_PIN			  16	// D0
    #define BUTTON1_PIN         5   //	D1
    #define BUTTON2_PIN         4	// D2
    #define BUTTON3_PIN         0	// D3
 //   Use the pin for the LED
 //   #define BUTTON4_PIN         2	// D4
    #define BUTTON5_PIN         14	// D5
    #define BUTTON6_PIN         12	//	D6
    #define BUTTON7_PIN         13	// D7
    #define BUTTON8_PIN         15	// D8
       

    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON4_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON5_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON6_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON7_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON8_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH

Valcob, 2018-10-19:
@moddsson Hey you said nothing about your button connections, are there some external pull-ups or not(your config explicitly says I'm gonna use for this button an external pull-up). So first thing, I tend to believe it's a hardware/software miss-configuration. In hardware.h there is a block where all the button modes are described:

> // BUTTON#_MODE: A mask of options (BUTTON_PUSHBUTTON and BUTTON_SWITCH cannot be together)
//   - BUTTON_PUSHBUTTON: button event is fired when released
//   - BUTTON_SWITCH: button event is fired when pressed or released
//   - BUTTON_DEFAULT_HIGH: there is a pull up in place
//   - BUTTON_SET_PULLUP: set pullup by software

And so happens that on D3 which is GPIO0 there is and external 10k pull-up this is why this one works as expected all other pins are in floating mode and can do what ever they want :) 
Hope this helps.

moddsson, 2018-10-19:
Thanks for the reply.

The setup I have at the moment is using the internal pullup (I think?  BUTTON_DEFAULT_HIGH ). And I simply short the corresponding pin to ground to simulate the button.  Your explanation makes sense, I will try to add extarnal pullps to the other pins and try again.  What confused me here was that the other pins worked properly for the first time and only the first time very consistently so it looked like a logic issue,  A floating condition might have represented itself more randomly.

xoseperez, 2018-12-09:
Your configuration (`BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH`) requires an external pull-up and GPIO0 has one that's why it works. You need to force the internal pull-up for the rest: `BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP`

moddsson, 2018-12-10:
Thanks for the response.  I can confirm that this is indeed the case.  I initially set the wrong button mode and got this strange "consistent" phenomenon where the pin worked once after each reboot.  After following the instructions everything works as intended.  Thanks for a great project!

xoseperez, 2018-12-10:
Great!

closed 2018-12-10 by xoseperez