xoseperez/espurna #1292: Quinled brightness vs MQTT vs HA

adrianmihalko, 2018-10-20:
First, thank you for making this great firmware. I want to use it with Quinled. I am happy to see it's supported.

There are small issue what I found:

- Brightness control is not added when I use auto discovery in Home Assistant, this part:

```
    brightness: true
    brightness_scale: 255
    brightness_command_topic: 'QUINLED-BED/brightness/set'
    brightness_state_topic: 'QUINLED-BED/brightness'
```

And I have one more question: what's the difference between setting brightness trough Brightness command vs. Channel #? Or I should leave both channels at maximum 255, then adjust brightness only with Brightness command?

Thank you,
Adrian

ColinShorts, 2018-10-20:
This might be related to #1222

adrianmihalko, 2018-10-21:
One more question/problem:

If I unplug espurna device, in HA device newer goes into unavailable state, and if was turned on it stay in state on. And this is problem. Is there any way to stay in sync?

1043717432, 2019-03-17:
> 正如你所写，有2个频道，＃0和[＃1](https://github.com/xoseperez/espurna/issues/1)：
> 
> [image]
> 
> 我想要的就是将两个通道都放在255（100％）上并使用亮度命令调整亮度（图片上的底部滑块）。
> 
> 这是不可能的？

This is my need. Have you solved the problem? Can you share the yaml file?

mcspr, 2019-03-18:
> All what I want is to leave both channels on 255 (100%) and adjust brightness with the Brightness command (bottom slider on the picture).
>
> This is not possible?

IMO the color check should just be removed. webui already exposes the slider and _generateBrightness handles it by setting all channels to the same value.

Not sure how second channel can be described through [HA mqtt.light](https://www.home-assistant.io/components/light.mqtt/) logic

copyrights, 2019-05-20:
BTW: <root>/brightness is already published, only subscription to <root>/brightness/set is missing

here is what mosquitto logs after mqtt.reset:
```
1558352737: Received UNSUBSCRIBE from ESPURNA-XXXXXX
1558352737: 	#
1558352737: ESPURNA-XXXXXX #
1558352737: Sending UNSUBACK to ESPURNA-XXXXXX
1558352737: Received SUBSCRIBE from ESPURNA-XXXXXX
1558352737: 	QuinLED2/channel/+/set (QoS 0)
1558352737: ESPURNA-XXXXXX 0 QuinLED2/channel/+/set
1558352737: Sending SUBACK to ESPURNA-XXXXXX
1558352737: Received SUBSCRIBE from ESPURNA-XXXXXX
1558352737: 	QuinLED2/relay/+/set (QoS 0)
1558352737: ESPURNA-XXXXXX 0 QuinLED2/relay/+/set
1558352737: Sending SUBACK to ESPURNA-XXXXXX
1558352737: Received SUBSCRIBE from ESPURNA-XXXXXX
1558352737: 	QuinLED2/pulse/+/set (QoS 0)
1558352737: ESPURNA-XXXXXX 0 QuinLED2/pulse/+/set
1558352737: Sending SUBACK to ESPURNA-XXXXXX
1558352737: Received SUBSCRIBE from ESPURNA-XXXXXX
1558352737: 	QuinLED2/led/+/set (QoS 0)
1558352737: ESPURNA-XXXXXX 0 QuinLED2/led/+/set
1558352737: Sending SUBACK to ESPURNA-XXXXXX
1558352737: Received SUBSCRIBE from ESPURNA-XXXXXX
1558352737: 	QuinLED2/action/set (QoS 0)
1558352737: ESPURNA-XXXXXX 0 QuinLED2/action/set
1558352737: Sending SUBACK to ESPURNA-XXXXXX
```

PR #1730 (2019-05-23): MQTT and HA brightness
MQTT and HA brightness shall only be depended on LIGHT_PROVIDER != LIGHT_PROVIDER_NONE

This should fix #1292. I did a quick test with MQTT and web interface on `intermittech-quinled` target without observations. The homeassistant part is untested.
closed 2019-05-23 by mcspr