xoseperez/espurna #1294: Disable REST API - Sonoff pow r2

wasiflaeeq, 2018-10-21:
Disabling REST API on espurna-1.13.3-itead-sonoff-pow-r2.bin works temporarily... 
When device is restarted it is enabled again... it probably does not save the values properly for this variable.

mcspr, 2018-10-23:
REST API meaning Restful? See https://github.com/xoseperez/espurna/issues/1280

Bochets34, 2018-11-03:
same issue with latest build(1.13.3) for SonOFF Dual R2, Restful API back to ON after power off/on cycle, or reboot, even if saved to Off before.

xoseperez, 2018-11-11:
I can't reproduce this. Maybe the issue is in the setting name as @mcspr suggests? The setting to disable the REST API from the web UI is the "Enable HTTP API" switch.

Bochets34, 2018-11-11:
unfortunately the issue is real, as by default with previous releases I 
was (and I'm still) using the HTTP API.

modes are not exclusive, so I can't see how enabling HTTP API could 
disable the REST one ?

anyway, I have 2 SonOff Dual R2 units with same behavior, upgraded to 
1.13.3, while my ~12 other are using an older release, and will stick to 
it until a fix appears.

I do have to put the REST API to off after a power off/power on cycle

mcspr, 2018-11-11:
Maybe rename 'Restful API' to 'Allow using GET to trigger HTTP API actions'?
Having both 'Enable HTTP API' and 'Restful API' is confusing may be interpreted as different things.

Overcrash86, 2018-11-17:
Same issue with sonoff basic.

Kaleidosko, 2018-11-23:
Same problem over here. Using Swa1 Lingan.

xoseperez, 2018-12-08:
@Bochets34 @Overcrash86 @Kaleidosko There was a problem in 1.13.3 with the "Restful API" switch fixed by @mcspr with #1288. Can you test this using `dev` branch?

closed 2018-12-11 by xoseperez