xoseperez/espurna #1298: ESP chip's hostname is not recognized or not in use

niniack, 2018-10-23:
On the WebUI, I have set the hostname to "studylamp" and when I save changes it shows up on the top left as well as in the configuration file through the following command: 

curl -u admin:mypassword http://myIPaddress/config

I have a simple lamp connected to my Sonoff S20 and I can toggle it using the following command without any issues: 

curl "http://myIPaddress/api/relay/0?apikey=XXXXXXXXXXXXXXX&value=2"

Similarly, I can copy the web address and paste this in my browser to have the same effect.

However, the issue arises when I switch out myIPaddress for studylamp.local 

Chromium throws me the following error: 

This site can’t be reached
studylamp.local’s server IP address could not be found.
DNS_PROBE_FINISHED_NXDOMAIN

AlbertWeterings, 2018-10-23:
On a Linux/Ubuntu machine "nslookup {IP}" gives you the full DNS name of your device but none of my ESPURNA devices is know by name in my network. So I think you are right and the hostname is not being send on the network.

I can't have a look in the code from my phone but hostname needs to be called before WiFi is started
WiFi.hostname("relay");
WiFi.begin(ssid, password);

mcspr, 2018-10-26:
Note that `.local` is mDNS, different from DNS names.

If you configure ESP with DHCP, your router may auto-assign DNS name to it (and if you are using router as dns server on the computer):
```
$ nslookup 10.0.0.2
2.0.0.10.in-addr.arpa       name = ESPURNA-123456.lan.
$ nslookup ESPURNA-123456.lan
Server:         10.0.0.1
Address:        10.0.0.1#53

Name:   ESPURNA-123456.lan
Address: 10.0.0.2 
```

AlbertWeterings, 2018-10-28:
Mcspr is right I tested configuring all ESP devices with DHCP and have them assign a IP based on MAC address by the DHCP. They all got a DNS name assigned that is the same as the device hostname.

closed 2018-12-09 by xoseperez