xoseperez/espurna #1300: "Too many messages queued" when using websockets debug

mcspr, 2018-10-23:
Thinking about https://github.com/xoseperez/espurna/commit/9f817ae2d0cc66bcbc0c683e9d51d28b19ee9c7a / #896 

Config difference from baseline:
pio platform **feature/stage** (see #1260 for ldscripts generator)
**esp8266/Arduino@5493552** / **2.4.2-85-ge5493552** / **lwip2** high bandwidth
https://github.com/me-no-dev/ESPAsyncTCP#7e9ed22
https://github.com/me-no-dev/ESPAsyncWebServer#bed4146
```c
#define WIFI_SLEEP_MODE WIFI_LIGHT_SLEEP
#define DEBUG_ESP_WIFI 1
#define ALEXA_SUPPORT 0         // to avoid network activity besides asyncwebserver
#define NTP_SUPPORT 0           // ^
#define SCHEDULER_SUPPORT 0     // ^
#define HOMEASSISTANT_SUPPORT 0 // ^
#define MQTT_SUPPORT 0          // ^
```

For example, using `keys`. `AsyncWebSocket.cpp` sections with os_printf / ets_printf are commented out.
Console output is filled with queue errors and yield does not really help here:
```
[580698] Current settings:
M: 32
W: 0 32
sent 32l bytes
S: 32 32
[580706] > adminPass => "fibonacci"
M: 43
[580709] > alexaEnabled => "0"
M: 38
[580711] > board => "2"
M: 31
[580713] > boardName => "NODEMCU_LOLIN"
M: 47
[580715] > btnGPIO0 => "0"
M: 34
[580717] > btnRelay0 => "0"
M: 35
[580718] > cfg => "3"
M: 29
[580720] > dns0 => "10.0.0.1"
M: 39
[580722] > gw0 => "10.0.0.1"
M: 40
ERROR: Too many messages queued
[580727] > hostname => "ESPURNA-35A259"
M: 47
ERROR: Too many messages queued
[580734] > ip0 => "10.0.0.2"
M: 40
ERROR: Too many messages queued
[580740] > ledGPIO0 => "2"
M: 34
...etc...
```

mcspr, 2018-10-24:
So I had done second pass at testing this, and it looks like this mostly affects Windows. For example, using Linux machine / Android phone throughput is much higher and filling of ws queue does not happen as quick (and bc it still happens, it should probably be slightly higher than 8 in this case)

See https://github.com/esp8266/Arduino/issues/1577

mcspr, 2018-10-27:
Adding to that, using `ets_printf()` actually *slows* everything down enough so .send() happens before queue fills up 🤕 Accidentally almost becoming yield.

Regarding delay, `info` memory consumption is notably less when using `delay(0)` instead of just `optimistic_yield()` (~2000byte data, mem is twice that + some). notably slower too, adding 6ms to every debugSend call instead of usual ~1-2ms.

tve, 2018-10-27:
Is there no way you can generate the data in the async callbacks of AsyncWebSocket?
In my experience with esp8266 TCP issues windows is worse at ACKing packets promptly than linux, but if your wifi is marginal then they can all suck. The wifi AP and any powersave settings can add to the fun. So unless you generate data in callbacks or have buffering for everything up-front you always risk overflowing. It's a PITA. BTW, I tend to use the low-memory version of LwIP (MSS of 536) 'cause it uses less memory so I can explicitly buffer more...

mcspr, 2018-10-28:
> Is there no way you can generate the data in the async callbacks of AsyncWebSocket?

Can you elaborate please? I am blanking here.

It sure does eat a lot memory with Windows when using current builds even when using OK WiFi, without sleep mode. Queue being unlimited adds to that and flashstrings malloc'ed twice in a row too.

I will try low-mem lwip version + buffer, that seems like a obvious way to avoid Windows assumptions about acking and allow sending more data. I wonder if cpu speed will affect this too.

mcspr, 2019-03-17:
Immediate issue fixed.
Next TODO:
fix-up ESPAsyncWebServer, avoid locking ws buffer queue
implement delivery status for each wsSend call, avoid thinking about ack's altogether (drop workaround).

closed 2019-03-17 by mcspr

zhivko, 2019-03-29:
Can you ceate pull request to ESPAsyncWebServer? I see similar issues.