xoseperez/espurna #1305: Does the firmware support GBLife RGB Socket?

etatus, 2018-10-31:
I recently adquired this smart socket that it's great because it has USB port and rgb light, but I can't find a custom firware that can support that.

As you can see in the picture, there is a specific board for the rgb lights, but it's still a ESP8266 to manage all the device.

[image]
[image]

ColinShorts, 2018-10-31:
Given that the device appears to show multiple colours at the same time, there's a good chance that it will need FastLED support to get those LED's working (possibly WS2811/2 or similar). If it only shows single colours at a time, then it's probably similar to (for example), the [Tonbux XS SSA06](https://github.com/xoseperez/espurna/wiki/Hardware-Tonbux-XS-SSA06)

Do you have any pictures of the front/back of the main boards - do the two boards separate easily?

I think @xoseperez is currently targetting [version 1.14.0](https://github.com/xoseperez/espurna/milestone/10) for FastLED support, but there are a number of outstanding issues to resolve, and it might not come in that version.

If the module is an ESP8266 then there is a reasonable chance you can support the basic relay functioning and power monitoring depending on the chip used.

etatus, 2018-11-01:
More pictures about the connection between the two boards.
[image]
[image]
[image]
[image]

ColinShorts, 2018-11-02:
Ok, it looks like the LEDs won't need fastLED. You'll just need to identify the best way to access the pins on the ESP for programming it, and then figure out which gpio pins are used to drive the relay, LEDs, power monitoring chip, and any switches.

You will probably need to separate the boards to attach wires unless you can see enough of the pads on the ESP module, or there are other points you can attach to. It'd probably be best to reattach the boards temporarily using longer wires until you've got it working.

I'd suggest trying your luck with the tonbux firmware and see if anything works, but you might need to change gpio assignment a few times, and you'll need to check the power chip, if fitted.

etatus, 2018-11-02:
Thank you @ColinShorts, I'll follow your advice. However, I'm not sure if the plug has power monitoring chip, how can I check that?

ColinShorts, 2018-11-02:
When you split the boards, take note of the chips and compare them against [the wiki](https://github.com/xoseperez/espurna/wiki/Sensors)

kerk1v, 2018-11-06:
Hello, 

if the led works and changes state, the GPIO is possibly correct. Probably you´re just missing 
```
    #define LED1_PIN_INVERSE	1
```
in hardware.h

Else, use a code block similar to 

```
    // LEDs
    #define LED1_PIN			13
    #define LED1_PIN_INVERSE	1
```

in your hardware.h to tinker with the GPIO port number.

etatus, 2018-11-06:
Thanks @kerk1v, that worked. Now, I have to figure out how to manage all the RGBW leds. I guess 4 GPIOs are needed and GPIOP0 seems to correspond to the red leds.

```
    // LEDs
    #define LED1_PIN            0   // R - 8 rgb led ring
    #define LED1_PIN_INVERSE    1
    #define LED2_PIN            5   // G
    #define LED2_PIN_INVERSE    1
    #define LED3_PIN            2   // B
    #define LED3_PIN_INVERSE    1
```

ColinShorts, 2018-11-06:
That's exactly how to do it, sadly, unless you're good at following traces, trial and error is the best way to do it. If you turn off the colour picker, you can more easily identify the individual pins - I went through the same process with the iWoole light 😁

etatus, 2018-11-07:
Ok, its done. I've figured out the RGBW pins to control the light and works like a charm with color picker. The problem is that when RELAY_PROVIDER is RELAY_PROVIDER_LIGHT, the socket RELAY does not work because current version of ESPurna doesn't allow a mixed of dummy and real relays (See https://github.com/xoseperez/espurna/issues/483).

@ColinShorts, @kerk1v, @xoseperez, any ideas to solve the problem?

```
    // Info
    #define MANUFACTURER        "TONBUX"  // This should be "GBLIFE"
    #define DEVICE              "XSSSA06"  // This should be "RGBW socket"

    // Buttons
    #define BUTTON1_PIN         13
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          15
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL
	
    // Light RGBW

    #define RELAY_PROVIDER      RELAY_PROVIDER_LIGHT
    #define LIGHT_PROVIDER      LIGHT_PROVIDER_DIMMER
    #define DUMMY_RELAY_COUNT   1
	
    #define LIGHT_CHANNELS      4
    #define LIGHT_CH1_PIN       5      // RED
    #define LIGHT_CH2_PIN       14      // GREEN
    #define LIGHT_CH3_PIN       12      // BLUE
    #define LIGHT_CH4_PIN       4      // WHITE
    #define LIGHT_CH1_INVERSE   0
    #define LIGHT_CH2_INVERSE   0
    #define LIGHT_CH3_INVERSE   0
    #define LIGHT_CH4_INVERSE   0	
```

etatus, 2018-11-09:
Ok, here is an alternative configuration with LED instead LIGHT, white LEDs are linked to the relay and red, green and blue LEDs can be controlled independently via MQTT. The light configuration is commented and will work when ESPurna allows a mixed of dummy and real relays (hopefully soon).

@xoseperez Maybe you want to include this configuration as a new supported hardwarde (I think it is the first that combines a rgbw light and a socket relay). In this issue there are also pictures of the hardware.

```
// -----------------------------------------------------------------------------
// GBLIFE RGBW SOCKET
// -----------------------------------------------------------------------------

#elif defined(GBLIFE_RGBW)

    // Info
    #define MANUFACTURER        "GBLIFE"
    #define DEVICE              "RGBW_SOCKET"

    // Buttons
    #define BUTTON1_PIN         13
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          15
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            4   // White LEDs
    #define LED1_PIN_INVERSE    0	
    #define LED1_RELAY          1

    #define LED2_PIN            5   // Red LEDs
    #define LED2_PIN_INVERSE    0
    #define LED2_MODE           LED_MODE_MQTT	
    #define LED3_PIN            14   // Green LEDs
    #define LED3_PIN_INVERSE    0   
    #define LED3_MODE           LED_MODE_MQTT
    #define LED4_PIN            12   // Blue LEDs
    #define LED4_PIN_INVERSE    0
    #define LED4_MODE           LED_MODE_MQTT	
 	
    // Light RGBW -- Valid when ESPurna supports a mixed of dummy and real relays

    //#define RELAY_PROVIDER      RELAY_PROVIDER_LIGHT
    //#define LIGHT_PROVIDER      LIGHT_PROVIDER_DIMMER
    //#define DUMMY_RELAY_COUNT   1
	
    // #define LIGHT_CHANNELS      4
    // #define LIGHT_CH1_PIN       5      // RED
    // #define LIGHT_CH2_PIN       14      // GREEN
    // #define LIGHT_CH3_PIN       12      // BLUE
    // #define LIGHT_CH4_PIN       4      // WHITE
    // #define LIGHT_CH1_INVERSE   0
    // #define LIGHT_CH2_INVERSE   0
    // #define LIGHT_CH3_INVERSE   0
    // #define LIGHT_CH4_INVERSE   0	
```

etatus, 2018-11-11:
@xoseperez I've found an easy way to make a dummy switch (for the light) and a real switch (for the relay) work **at the same time**. It is necessary to make a few changes in the source code:

In **relay.ino** comment #else to force applying real relay configuration (if exists):
```c
void relaySetup() {

    // Dummy relays for AI Light, Magic Home LED Controller, H801,
    // Sonoff Dual and Sonoff RF Bridge
    #if DUMMY_RELAY_COUNT > 0

        // No delay_on or off for these devices to easily allow having more than
        // 8 channels. This behaviour will be recovered with v2.
        for (unsigned char i=0; i < DUMMY_RELAY_COUNT; i++) {
            _relays.push_back((relay_t) {0, RELAY_TYPE_NORMAL, 0, 0, 0});
        }

    //#else   <----------

        #if RELAY1_PIN != GPIO_NONE
            _relays.push_back((relay_t) { RELAY1_PIN, RELAY1_TYPE, RELAY1_RESET_PIN, RELAY1_DELAY_ON, RELAY1_DELAY_OFF });
        #endif

        ....
```

In **relay.ino** replace in __relayProviderStatus_ function:
`#if RELAY_PROVIDER == RELAY_PROVIDER_LIGHT`
to
`#if DUMMY_RELAY_PROVIDER == RELAY_PROVIDER_LIGHT`

In **hardware.h** include button, relay, light and optionally led configuration for the device. Switch #0
(DUMMY_RELAY_PROVIDER) activates the light and Switch #1 (GPIO15) activates the relay.
```c
// -----------------------------------------------------------------------------
// GBLIFE RGBW SOCKET
// -----------------------------------------------------------------------------

#elif defined(GBLIFE_RGBW)

    // Info
    #define MANUFACTURER        "GBLIFE"
    #define DEVICE              "RGBW_SOCKET"

    // Buttons
    #define BUTTON1_PIN         13
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       2

    // Relays
    #define RELAY2_PIN          15
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL 
 
    // LEDs
    #define LED1_PIN            4   // W
    #define LED1_PIN_INVERSE    0
    #define LED1_MODE           LED_MODE_MQTT	
    //#define LED1_RELAY          2

    #define LED2_PIN            5   // R - 8 rgb led ring
    #define LED2_PIN_INVERSE    0
    #define LED2_MODE           LED_MODE_MQTT	
    #define LED3_PIN            14   // G
    #define LED3_PIN_INVERSE    0   
    #define LED3_MODE           LED_MODE_MQTT
    #define LED4_PIN            12   // B
    #define LED4_PIN_INVERSE    0
    #define LED4_MODE           LED_MODE_MQTT	 
 
 	
    // Light RGBW 

   // ***** Changed RELAY_PROVIDER to DUMMY_RELAY_PROVIDER *****
    #define DUMMY_RELAY_PROVIDER RELAY_PROVIDER_LIGHT

    #define LIGHT_PROVIDER       LIGHT_PROVIDER_DIMMER
    #define DUMMY_RELAY_COUNT   1
	
    #define LIGHT_CHANNELS      4
    #define LIGHT_CH1_PIN       5      // RED
    #define LIGHT_CH2_PIN       14      // GREEN
    #define LIGHT_CH3_PIN       12      // BLUE
    #define LIGHT_CH4_PIN       4      // WHITE
    #define LIGHT_CH5_PIN       2      // DARKNESS
    #define LIGHT_CH1_INVERSE   0
    #define LIGHT_CH2_INVERSE   0
    #define LIGHT_CH3_INVERSE   0
    #define LIGHT_CH4_INVERSE   0	
    #define LIGHT_CH5_INVERSE   0
```

xoseperez, 2018-12-09:
@etatus, thanks! Used part of your code but used a simpler approach to allow physical relays on a light device. Can you check if it's working?

xoseperez, 2018-12-10:
Also changed the order the relays are loaded. Now physical relays are loaded first and dummy relays afterward. I have also changed the device configuration to reflect this.

etatus, 2018-12-10:
Great news @xoseperez! I'll give a try and I'll let you know. Do I have to change any particular thing in my hardware.h? Did you preserve DUMMY_RELAY_PROVIDER variable?

xoseperez, 2018-12-10:
The configuration is in `hardware.h`. I'm not using the DUMMT_RELAY_PROVIDER setting.

etatus, 2018-12-10:
@xoseperez I confirm the changes work properly. I had to make some changes in my configuration (physical relays go first now).

Just one issue to fix: when I switch physical relay in Home Assistant, it is updated in the ESPURNA web interface (Status), but when I switch dummy relay in Home Assistant, it is not updated in the ESPURNA web interface, despite it works fine. I guess it is a problem with MQTT communication.

And a final quesiton: Is it possible to get the status of normal LED sensors? You can turn them on/off, but you cannot know if they are turned on/off.

```c
// -----------------------------------------------------------------------------
// GBLIFE RGBW SOCKET
// -----------------------------------------------------------------------------

#elif defined(GBLIFE_RGBW)

    // Info
    #define MANUFACTURER        "GBLIFE"
    #define DEVICE              "RGBW_SOCKET"

    // Buttons
    #define BUTTON1_PIN         13
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1

    // Relays
    #define RELAY1_PIN          15
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL 
 
    // LEDs  <--- This LED configuration is optional
    #define LED1_PIN            4   // WHITE  - 8 rgb led ring
    #define LED1_PIN_INVERSE    0
    #define LED1_MODE           LED_MODE_MQTT	
    //#define LED1_RELAY          1
    #define LED2_PIN            5   // RED
    #define LED2_PIN_INVERSE    0
    #define LED2_MODE           LED_MODE_MQTT	
    #define LED3_PIN            14   // GREEN
    #define LED3_PIN_INVERSE    0   
    #define LED3_MODE           LED_MODE_MQTT
    #define LED4_PIN            12   // BLUE
    #define LED4_PIN_INVERSE    0
    #define LED4_MODE           LED_MODE_MQTT	 
 
 	
    // Light RGBW 

    #define RELAY_PROVIDER      RELAY_PROVIDER_LIGHT
    #define LIGHT_PROVIDER      LIGHT_PROVIDER_DIMMER
    #define DUMMY_RELAY_COUNT   1
	
    #define LIGHT_CHANNELS      4
    #define LIGHT_CH1_PIN       5      // RED
    #define LIGHT_CH2_PIN       14      // GREEN
    #define LIGHT_CH3_PIN       12      // BLUE
    #define LIGHT_CH4_PIN       4      // WHITE
    #define LIGHT_CH5_PIN       2      // DARKNESS
    #define LIGHT_CH1_INVERSE   0
    #define LIGHT_CH2_INVERSE   0
    #define LIGHT_CH3_INVERSE   0
    #define LIGHT_CH4_INVERSE   0	
    #define LIGHT_CH5_INVERSE   0
```

xoseperez, 2018-12-10:
Why do you have both LEDs and channels to the same GPIOs? And what does the "darkness" channel do? I thought it was a 4 channel light

xoseperez, 2018-12-11:
You can also access individual light channels over MQTT under this kind of topic: `ESPURNA_XXXXXX/channel/0/set`. You can even define 4 dummy relays and each one will control one of the channels individually (or 5 dummy relays and the first one will control all channels and the others one each).

About the darkness channel, it's absurd, indeed.

etatus, 2018-12-11:
Ok, I guess it's a better approach to use channels topic intead leds.

BTW I read somewhere the darkness channel could refer to color saturation level.

xoseperez, 2018-12-11:
Mmm.. interesting. ESPurna does not have any way to specify the saturation only at the moment.

closed 2018-12-11 by xoseperez