xoseperez/espurna #1308: SONOFF_RFBRIDGE: second RF code off code is copied over to all RF codes after number 10

vogelfreiheit, 2018-11-01:
In case the title is confusing:

1. Adding the code tot he off state of the 2nd RF "relay" results in the same code being replicated/set to every relay starting at #10 when support for more than 9 relays is set at compile time.

https://imgur.com/a/QVLVHC2

mdmbc, 2018-11-13:
Can confirm.. Same happens for me.

mcspr, 2018-11-15:
https://github.com/xoseperez/espurna/blob/a89b511f38b7f614a4873b7616635e6863e2d263/code/espurna/rfbridge.ino#L531-L535
by the looks of it, key buffer buffer is too small and needs to be 9 instead of 8
e.g. `rfbOFF10`, `rfbOFF11` ... becomes `rfbOFF1`

mcspr, 2018-11-15:
@vogelfreiheit can you test changes from #1348?

closed 2018-11-19 by xoseperez

vogelfreiheit, 2018-11-20:
Will test ASAP and comment! Thanks Xose for all the good work.

PR #1348 (2018-11-19): Proper buffer size to fit two digit rfbOFF key
Resolves #1308