xoseperez/espurna #1309: Encendido automatico no programado

adolfossl, 2018-11-01:
Hola!!!
He detectado en la Nightly build (20181011) que cuando se activa un schedule cualquiera, además el evento programado también se ejecuta un encendido a la 01:30 y otro sobre las 04:00. Esto no ocurre si no hay ningún schedule activo.
Lo he comprobado con dos Sonoff, el S26 y el basic.

Gracias

Valcob, 2018-11-06:
@adolfossl  Hi, so basically you are saying that you have a schedule that is fired twice once at 1.30 and another at 4 for no reason, right? Please keep English let's respect each other so anyone can understand what you wrote, thank you in advance. Also could you please try the latest stable build not the nightly one and try reproduce the issue.

xoseperez, 2018-11-11:
Hi @adolfossl 
Are you still experiencing this issue?
Can you check if there is something else that could cause that behavior? Like something in your network triggering the switch or maybe the board resetting with an "always on" setting on boot?

adolfossl, 2018-11-13:
Nothing is swithing the sonoff, the stable version is working ok

closed 2019-02-16 by stale[bot]