xoseperez/espurna #1310: Add option to initialize time to given date when booting

skorokithakis, 2018-11-01:
I created an Espurna-based dog feeder for my house, and use schedules to turn it on once a day. Unfortunately, the wifi connection to the Sonoff is a bit spotty, so I'm afraid that there might be a power failure and the Sonoff won't be able to re-establish internet for a while, thus the schedules won't work.

Would it be possible to either initialize the time to a given date when booting, e.g. 2018-01-01 12:00:00 so I can know that schedules will work (obviously shifted by N hours, but at least they'll trigger) or have schedules that trigger every X hours based on an internal timer and not the RTC?

poulch74, 2018-11-01:
in fact time service based on millis() and sync with NTP periodically. with some coding you can set any time but it be innacurate.  i make some pull requests in rtc branch (see repo), so you can attach cheap i2c ds3231 rtc module and have accurate local time without internet connection.

skorokithakis, 2018-11-01:
I tried that but it ended up setting the same date whenever it tried to connect to the NTP server. Do you know where I can change the initial date so it can be overridden by NTP?

poulch74, 2018-11-02:
in my rtc branch i provide alter sync routine for time lib. See call setSyncProvider(ntp_getTime); in ntp.cpp and rtc.cpp.  I think you can try same thing - change SynProvider then ntp fail and provide fake time...

skorokithakis, 2018-11-02:
I'll look into it, thanks, although providing fake time on NTP fail isn't the right solution because it would just always overwrite the current time, even if it had synced previously. I'll figure something out, though.

poulch74, 2018-11-02:
you can make global static flag variable to store success NTP sync attempts... if flag zero - provide fake time, else your device have some valid inaccurate time.

xoseperez, 2018-12-09:
I just added a small build-time option to allow the scheduler to check the time even if there has not been am NTP sync yet. You just have to change the `NTP_WAIT_FOR_SYNC` to 0 in `general.h` (or in a `custom.h` file or as a command line setting when building using platformio).

By default, time is set to the origin of UNIX timestamp, i.e. 1970-01-01 00:00:00.

skorokithakis, 2018-12-10:
This is great, thank you! It should work beautifully for my use case, and I never considered such a simple implementation, thanks again!

closed 2018-12-11 by xoseperez