xoseperez/espurna #1311: add support for prodino wifi esp

dpeddi, 2018-11-02:
I've a prodino wifi esp. I'm interested in get it supported.

Since the device have onboard an MCP23S08 to manage input and output, I have to wait for virtual gpio support.

Regards

Valcob, 2018-11-02:
@dpeddi Yes, that is a planned thing please be patient. All buttons, leds, relays, lights will have a functionality provider instead of a direct GPIO access so it's a pinned feature that will be in the next major releases. Currently there is a prototype on STM8 as GPIO expander (SERIAL communication) but it is a WIP so it's comming.

dpeddi, 2018-11-02:
Great, thank you!

xoseperez, 2018-11-11:
See also #199

dpeddi, 2019-11-10:
@xoseperez any progress about the functionality provider? There is a more descriptive example? 
Regards

mcspr, 2019-11-12:
If you mean code example, perhaps something like this:
https://github.com/davetcc/IoAbstraction/blob/master/src/IoAbstractionWire.cpp
https://github.com/esphome/esphome/blob/75275c4e934746f33f81f6cc4aea251d04e88783/esphome/components/mcp23017/mcp23017.cpp
https://github.com/esphome/esphome/blob/75275c4e934746f33f81f6cc4aea251d04e88783/esphome/components/pcf8574/pcf8574.cpp
We just need to replace `digitalWrite` / `digitalRead` / `pidMode` with some kind of proxy to work with either real pins or a board like this.

If I remember this right, there was a prototype work mentioned somewhere, but it was never published.

dpeddi, 2020-06-01:
I've spent some time and i've completed the relay support for this board.
Now I'm implementing the opto input.

I plan to pull request within a couple of day.

dpeddi, 2020-06-03:
I need some help for extending correctly DebounceEvent... I hope you could help me

closed 2020-07-16 by mcspr

ardupic, 2020-11-21:
hi,
thanks to @mcspr  for build i found [there](https://github.com/mcspr/espurna-nightly-builder/releases/download/20201103/espurna-1.15.0-dev.nightly20201103+git289c8cb4-prodino-wifi.bin)

it works great.. but  i m looking for a solution:
digital inputs/Button are latched to their relay... i think  with `#define BUTTON1_RELAY           1`
i need to unbind inputs and outputs and only control and aquire states of them via MQTT
can you help me to do this in a simply way?
thanks

mcspr, 2020-11-21:
@ardupic this is not part of documentation, yet, as this was meant to be included in the webui initially...
https://github.com/xoseperez/espurna/blob/850db61b761ebb8a4496bb031922f091b6df563e/code/espurna/button.cpp#L835
https://github.com/xoseperez/espurna/blob/850db61b761ebb8a4496bb031922f091b6df563e/code/espurna/config/types.h#L418
Settings work as an overlay on top of the build flags, adding `btnRelay0` set to `153` will override the existing flag without need to rebuild

https://github.com/xoseperez/espurna/wiki/Buttons-and-switches should probably include these keys though

ardupic, 2020-11-22:
thanks @mcspr,
I think to not be able to set all the correct  '#Define .... ' (arduino` IDE) in espurna code to make it compatible with prodino wifi and do also changes in your last post. :((    
Are you going to implement espurna settings web page  for this borad?, so i can easly change button settings.
I have several prodino wifi i would exploit with this firmware. could i wish to see full integration in espurna?

thk a lot

mcspr, 2020-11-22:
@ardupic no need to rebuild. The comment above meant to point to these settings, which you can set via DEBUG panel:
```
set btnRelay0 159
set btnRelay1 159
set btnRelay2 159
set btnRelay3 159
```
To delete, `del btnRelay0` or the other key you want.

Meaning, for the 1st button the code does:
- Read `btnRelay0` setting. When not found, do the next step
- Read `BUTTON1_RELAY` build flag. 
- Read `BUTTON1_RELAY` build flag from defaults.h

Button number `0` in settings means `1` button in the build, `1` is `2` and etc.

ardupic, 2020-11-24:
hi, @mcspr,
thank for yor suggestion but i still have problem.
in order:
if i send `set btnRelay0 159`  or `set btnRelay0 153` nothing changes and console monitor return me 
`
[279587] [WEBSOCKET] Requested action: dbgcmd
-ERROR: Command not found
`
some times returm me an `Ok` but no cange take effect

an other stange behaivor is:
if i hardware short press my inputs, realys toggle normally but if i long press  them, board looses connection and browser ask me to reload the page
`[525691] [BUTTON] Button #0 event 1 (pressed)
[526196] [BUTTON] Button #0 event 3 (click)
[526198] [MQTT] Sending lavatrice/button/0 => click (PID 1)
[526199] [RELAY] #0 scheduled OFF in 0 ms
[526212] [RELAY] #0 set to OFF
[526213] [MQTT] Sending lavatrice/relay/0 => 0 (PID 1)
[527216] [RELAY] Setting relay mask: 0b1110
[530660] [BUTTON] Button #0 event 1 (pressed)
[533209] [BUTTON] Button #0 event 5 (long-click)
[533211] [MQTT] Sending lavatrice/button/0 => long-click (PID 1)
`

mcspr, 2020-11-24:
Setting will be read on the next reboot. Number does not matter (...atm, might change), it's just some value that is out-of-range.

Default `long-click` action on the 1st button is to reboot the device:
https://github.com/xoseperez/espurna/blob/850db61b761ebb8a4496bb031922f091b6df563e/code/espurna/config/defaults.h#L187
Although, log should probably show the info about the action taken.

Actions are controlled through settings as well:
https://github.com/xoseperez/espurna/blob/850db61b761ebb8a4496bb031922f091b6df563e/code/espurna/button.cpp#L846-L852
Where action IDs are listed in types header:
https://github.com/xoseperez/espurna/blob/850db61b761ebb8a4496bb031922f091b6df563e/code/espurna/config/types.h#L23-L36

ardupic, 2020-11-24:
ok, something is working.
buttons and relay are indipendent now.
there only a few issues, long press of button 0 reboot the board.
in mqtt if i subscribe "lavatrice/button/1" for example, the only payload avavailable is "click" that i got it only on release of hardware button

ardupic, 2020-11-24:
dear   @mcspr    your post are helpfull but
:(((((
last update... changing settings everywhere, looks like working enough.
i deleted th action to prevent reboot 
all channells send the mqtt event "pressed" but 0 an 1 channel don't send mqtt message on release, buttons 2 and 3 on release send only "click" on short press and nothing on long press
finally i would receive in mqtt for each button/input a payload like "pressed" or "on"  when each channel is on  and a message like "released" when channells are off.
thanks
p.s. have you a list of debug consolle commands?

mcspr, 2020-11-25:
@ardupic see https://github.com/xoseperez/espurna/wiki/Buttons-and-switches#mqtt and https://github.com/xoseperez/espurna/wiki/Terminal

ardupic, 2020-11-25:
@mcspr thanks for all your help and upgrading the wiki
i finally solved finding in espurna code this command "btnMode"...
maybe here https://github.com/xoseperez/espurna/blob/dev/code/espurna/button.cpp
i've tried to use via debug consolle and magically trasfoms my inputs from button to switch and than mqtt publish only a pressed and a release payload, no click or long click  payloads that are not usefull for me.
this is the command i've used
set btnMode# 1 where # is input index

PR #2269 (2020-07-16): Implement support for prodino wifi esp
As promised, here is a pull request to implement basic support for prodino wifi esp (fix #1311)

Some of the code for the Expander is borrowed by the original board developer (I left them in the copyright statement)

The relay works great.

The opto works as well, but since I connected them to some button, I have some bounce.

I think you can accept this request, and later we could move the expander poll routine in DebounceEvent