xoseperez/espurna #1312: Add delay to the connection

Jaimillo, 2018-11-02:
Greetings to all.

I am starting to use Espurna with Sonoff devices (POW R2) and my intention is to connect them to several household appliances like the Washing machine, the air conditioning equipment, etc.
In these household appliances that I mentioned, for security I always program in the "SWITCHES" menu the BOOT mode "Always OFF", but in the device that I want to connect to the refrigerator, I will have to put "Always ON".
The question I have is whether it is currently possible (or it may be an improvement for future versions) to program a delay for the relay to connect after a time since the Sonoff device receives alternating current (for example 30 seconds).
The reason for this pause is to protect the refrigerator (and other equipment supplied to the power grid) from the voltage spikes that occur when the power returns, as well as the possible relapses of the network, until it stabilizes.

Using MQTT, I could solve it, but in the event of the fall of the electrical network, I am not sure that the server, the Wi-Fi network, etc. will continue to work.

Greetings.

xoseperez, 2018-11-11:
You can configure that behavior if you set the device connected to the fridge as "always off" on boot but with a "normally on" pulse mode with a 30s pulse time.

Jaimillo, 2018-11-11:
Thank you very much for the information Xoxe, this afternoon I try it.
Greetings.

Jaimillo, 2018-11-11:
Tested and works correctly as you said, thank you very much.
Now it is the most difficult, convince the parienta, to let me install it in the refrigerator ;)
Greetings.

xoseperez, 2018-12-13:
I'm sure your parienta will appreciate it when you contact her from 1000km away to tell her she left the fridge door open. It happened to me :smile:

closed 2018-12-13 by xoseperez