xoseperez/espurna #1315: How to use more than one SHT31

flashy02, 2018-11-03:
Hi, i have a Problem to use two SHT31 at one I2C port. Because it is possible to use these sensors at Adresses 0x44 and 0x45 it should be possible to use two sensor at one I2C. How have I to define this in hardware.h? My actual definition doesn't work:

    // A bit of I2C - pins 4,5
    #define I2C_SDA_PIN         4
    #define I2C_SCL_PIN         5

    // Sensors!
    #ifndef SENSOR_SUPPORT
    #define SENSOR_SUPPORT     1
    #endif
    #ifndef SHT3X_I2C_SUPPORT
    #define SHT3X_I2C_SUPPORT     1
    #endif
    #define SHT3X_I2C_ADDRESS    0x44
    #define SHT3X_I2C_ADDRESS    0x45

mcspr, 2018-11-11:
Hello.
You can try adding second sensor instance, like here https://github.com/xoseperez/espurna/issues/934

Also note that in your example defining `SHT3X_I2C_ADDRESS` second time overwrites previous value.
(if you are using ArduinoIDE, check out it's settings via <kbd>Ctrl</kbd>+<kbd>,</kbd> and look for compiler warnings / verbose output. you will see warnings that way)

flashy02, 2018-11-12:
Thank you for the hint, I'll give it a try.

closed 2018-12-12 by xoseperez