xoseperez/espurna #1316: sonoff bridge rf 433 first init problem

lcdkk, 2018-11-03:
hi;

when i was flash the new sonoff bridge rf , and then find espruna ssid, and login see the
change admin password screen, and generate the new password then save,
but now lost connection wifi, and unable to login.

any on can assist this ?

thanks.
neo

mcspr, 2018-11-06:
Use the new password. It is shared between web and wifi.

pit001, 2018-11-11:
Here also. 
This is new (not with FW 12) and surpricing  - I also need 3 reflashing loops.

seetendra, 2019-01-01:
Is there a way by which I can build a bin file with my Wifi Password integrated into it.

lucaberta, 2019-01-01:
> Is there a way by which I can build a bin file with my Wifi Password integrated into it.

@seetendra look for the default wifi credentials in general.h:

https://github.com/xoseperez/espurna/blob/dev/code/espurna/config/general.h

seetendra, 2019-01-01:
@lucaberta Thanks a lot for your reply.
Can you please help me with the Sonoff POW as well. If you have tried Tasmota it has the option to look for the Total Power consumption (Sonoff POW) and Daily Consumption as well.
Is there a Way we can see that with Espurna ?

lucaberta, 2019-01-01:
> @lucaberta Thanks a lot for your reply.
> Can you please help me with the Sonoff POW as well. If you have tried Tasmota it has the option to look for the Total Power consumption (Sonoff POW) and Daily Consumption as well.
> Is there a Way we can see that with Espurna ?

@seetendra you might want to direct this sort of questions to the Gitter channel dedicated to discussing ESPurna like Xose says in the README file:

> Please use the [gitter ESPurna channel](https://gitter.im/tinkerman-cat/espurna) for support and questions, you have better chances to get fast answers from me or other ESPurna users. Open an issue here only if you feel there is a bug or you want to request an enhancement. Thank you.

seetendra, 2019-01-01:
Ok will do

mcspr, 2019-02-26:
Initial screen should have ssid/pass entries for a single network. OSes make it non-trivial to instantly forget the network ssid when it changes. Windows 10, for example, will refuse connection without any warning and requires manually "forgetting" the network before second connection by closing-opening wifi listing.

This also can be mitigated by removing wifi settings reload, which restarts AP immediately when user changes password. WebUI can then reload and ask new password, allowing to change settings as usual. Next settings reload can drop the connection

closed 2019-07-10 by stale[bot]