xoseperez/espurna #1317: Send RAW IR commands issues

izharm, 2018-11-03:
I have tested this feature in version 1.13.3a and found the following issues:

1. Raw Rx doesn't work, _irRXLoop() fails on "check code" and returns. I was able to get it to work by moving the checks :

        // Check code
        if (_ir_results.value < 1) return;
        if (_ir_results.decode_type < 1) return;
        if (_ir_results.bits < 1) return;

to the none ir_use_raw section.

I didn't test this with the non raw option though.

2. IR Tx MQTT messages were split on reception and the IR send code wasn't working well.  It seems this is due to TCP buffer size, as described here [https://github.com/marvinroger/async-mqtt-client/issues/98#issuecomment-379969767](url), so changing the lwIP (lightweight IP) option to "V2 higher bandwidth" solved this issue.

I am now able to receive and send raw IR message.

By the way, the code mentions to "increase MQTT packet size -DMQTT_MAX_PACKET_SIZE=1200" - As far as I can tell, this doesn't work when using MQTT_USE_ASYNC, maybe you need to add _mqtt.setMaxTopicLength in _mqttConnect().  I didn't see a difference and the code worked either way (once the TCP issue was resolved) 

thanks for your continuing efforts.

Izhar

_Originally posted by @izharm in https://github.com/xoseperez/espurna/issues/556#issuecomment-427575844_

vtochq, 2018-11-07:
I test it right now and it works without moving checks to non-raw section. Maybe codes from my remote decoded by library.
But I agree, those checks not for raw option. I make pull request.

Yes, MQTT_MAX_PACKET_SIZE used only by PubSubClient.
It's my mention and my mistake. Maybe I debug IR Raw with PubSub... Don't remember.

mcspr, 2018-11-11:
Resolved by https://github.com/xoseperez/espurna/pull/1324?

xoseperez, 2018-12-11:
@izharm Can you confirm @vtochq changes fix the issue?

izharm, 2018-12-12:
Hi,

I'm going to need a few days to get this setup up and running again.  I'l
let you know once it's tested.

Izhar

closed 2018-12-21 by xoseperez