xoseperez/espurna #1320: Sonoff POW R2 serial console not working

lucaberta, 2018-11-05:
Just received two POW R2 unit, and after flashing the latest build as downloaded from GitHub, it looks like there is absolutely *NO* console output beyond the ESP8266 boot log:

```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
```
then nothing more gets printed on the serial console port. Not a single character.

The web interface works regularly, and thru it I was able to configure the device and turn on the telnet interface, from which I could download the following information:

```
[757175] [MAIN] ESPURNA 1.13.3 (e2641b9)
[757178] [MAIN] xose.perez@gmail.com
[757181] [MAIN] http://tinkerman.cat

[757185] [MAIN] CPU chip ID: 0x3234A5
[757188] [MAIN] CPU frequency: 80 MHz
[757191] [MAIN] SDK version: 1.5.3(aec24ac9)
[757194] [MAIN] Core version: 2.3.0
[757198] [MAIN] Core revision: 159542381
[757201] 
[757204] [MAIN] Flash chip ID: 0x1640EF
[757210] [MAIN] Flash speed: 40000000 Hz
[757213] [MAIN] Flash mode: DOUT
[757214] 
[757217] [MAIN] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[757223] [MAIN] Flash size (SDK)    :  4194304 bytes / 1024 sectors (   0 to 1023)
[757226] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[757230] [MAIN] Firmware size       :   480416 bytes /  118 sectors (   1 to  118)
[757233] [MAIN] Max OTA size        :   540672 bytes /  132 sectors ( 119 to  250)
[757237] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[757239] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[757240] 
[757243] [MAIN] EEPROM sectors: 1019, 1018, 1017, 1016
[757246] [MAIN] EEPROM current: 1017
[757249] 
[757253] [MAIN] EEPROM:  4096 bytes initially |  1281 bytes used (31%) |  2815 bytes free (68%)
[757258] [MAIN] Heap  : 35912 bytes initially | 20752 bytes used (57%) | 15160 bytes free (42%)
[757261] [MAIN] Stack :  4096 bytes initially |  1352 bytes used (33%) |  2744 bytes free (66%)
[757264] 
[757268] [MAIN] Boot version: 7
[757271] [MAIN] Boot mode: 1
[757274] [MAIN] Last reset reason: Reboot from web interface
[757277] 
[757281] [MAIN] Board: ITEAD_SONOFF_POW_R2
[757284] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB 
[757288] [MAIN] Sensors: CSE7766 
[757292] [MAIN] WebUI image: SENSOR
[757295] 
[757298] [MAIN] Firmware MD5: a2bbf3bb4f80936582ce98c5c90aca77
[757302] [MAIN] Power: 3006 mV
[757306] [MAIN] Power saving delay value: 10 ms
[757309] 
```

The unit seems to be working fine, and once the box is closed again, this lack of serial console is not going to have any impact whatsoever, still this is a nuisance as the serial console.

I can compile new code and upload it to the unit in case some debugging or a test build be created, to test it.

Thanks,

Luca

mcspr, 2018-11-06:
[Hardware definition explicitly disables debug output over serial (`DEBUG_SERIAL_SUPPORT`)](https://github.com/xoseperez/espurna/blob/88a7076ad3b906f8aeee0ae6978332f0e61b4335/code/espurna/config/hardware.h#L442-L470)

Power measuring chip (CSE7766) is talking to ESP via serial.

JohnOH, 2018-11-06:
The serial GND is also connected to AC live. Surprising you were able to receive what you did if you were AC connected.

lucaberta, 2018-11-06:
>[Hardware definition explicitly disables debug output over serial (`DEBUG_SERIAL_SUPPORT`)](https://github.com/xoseperez/espurna/blob/88a7076ad3b906f8aeee0ae6978332f0e61b4335/code/espurna/config/hardware.h#L442-L470) ...

@mcspr this is what happens when one is keen to get Espurna up and running on some new devices, and it’s midnight or so... reading the wiki, naaah, maybe tomorrow! 😄

Thanks for the clarification on this major difference between POW and POW R2.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

lucaberta, 2018-11-06:
>The serial GND is also connected to AC live. Surprising you were able to receive what you did if you were AC connected. 

@JohnOH I only do **that** kind of mistake once! Last week I fried a Sonoff TH10, a CP2102 USB adapter, and one USB port on the PC as I was curious to debug the serial port on an AC-connected device. Then then magic smoke came... 😬

Last night I has just flashed the POW R2 binary on the serial console with esptool.py and was baffled when the Atom terminal was not showing anything after the few lines of the ESP8266 boot.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2018-11-06 by lucaberta