xoseperez/espurna #1325: Upgrade to Gulp 4

Niek, 2018-11-08:
Gulp 3.9 is used now, but is has some dependency vulnerabilities and it doesn't work with Node 10 (https://github.com/gulpjs/gulp/issues/2162). Upgrading to Gulp 4 would require some changes in the gulpfile.

mcspr, 2018-11-08:
True. Although this might be the case of old `natives` version.
It works with 1.1.6, but not with 1.1.1 (locked by package-lock.json):
```
builder:espurna-test/code $ npm ls natives
esp8266-filesystem-builder@0.2.2 espurna/code
├─┬ gulp@3.9.1
│ └─┬ vinyl-fs@0.3.14
│   └─┬ graceful-fs@3.0.11
│     └── natives@1.1.6  deduped
└── natives@1.1.6

builder:espurna-test/code $ gulp webui_small
[14:12:28] Using gulpfile ~/espurna-test/code/gulpfile.js
[14:12:31] Starting 'webui_small'...
Image index.small.html.gz       size: 50653 bytes
[14:12:35] Finished 'webui_small' after 3.5 s
builder:espurna-test/code $ npm install 'natives@1.1.1'
npm WARN deprecated natives@1.1.1: This module relies on Node.js's internals and will break at some point. Do not use it, and update to graceful-fs@4.x.
npm WARN esp8266-filesystem-builder@0.2.2 No repository field.

+ natives@1.1.1
updated 1 package and audited 1911 packages in 8.541s
found 10 vulnerabilities (1 low, 5 moderate, 4 high)
  run `npm audit fix` to fix them, or `npm audit` for details

builder:espurna-test/code $ gulp webui_small
gulp[2939]: ../src/node_contextify.cc:628:static void node::contextify::ContextifyScript::New(const v8::FunctionCallbackInfo<v8::Value>&): Assertion `args[1]->IsString()' failed.
 1: 0x8daaa0 node::Abort() [gulp]
 2: 0x8dab75  [gulp]
 3: 0x90e2f7 node::contextify::ContextifyScript::New(v8::FunctionCallbackInfo<v8::Value> const&) [gulp]
 4: 0xb5e71b  [gulp]
 5: 0xb606b2 v8::internal::Builtin_HandleApiCall(int, v8::internal::Object**, v8::internal::Isolate*) [gulp]
 6: 0x3443a2b5be1d
Aborted (core dumped)
```

Niek, 2018-11-08:
Yes, if you use yarn instead of NPM, it can be fixed by adding this to the package.json:
```js
  "resolutions": {
    "natives": "1.1.3" // Required for Node 10 support (only works with yarn)
  },
```
But that's more like a temporary workaround, it breaks again in Node 11 for example.

PR #1403 (2018-12-09): Update gulp@4
Resolves #1325 

My only question is pinned "map-stream": "0.0.7" from 7b63521c80a37e854fc8941abb0aa21a48f60b86
closed 2018-12-09 by xoseperez