xoseperez/espurna #1329: Arduino IDE ESPAsyncTCP.cpp compile error

luisgcu, 2018-11-10:
Hello,
I am new trying to use the spurna project and I got a lot error because missing libraries, I installed missing libraries , but there is on error that I don't know how to solve that.
ESPAsyncTCP\src\ESPAsyncTCP.cpp:113:36: error: macro "ip_route" requires 2 arguments, but only 1 given

   netif* interface = ip_route(&addr);

AlbertWeterings, 2018-11-10:
Can you try this lib might help to manually remove any old version of the lib
https://github.com/me-no-dev/ESPAsyncTCP#55cd520

luisgcu, 2018-11-10:
thanks @AlbertWeterings  problem solved!!, I am pretty sure that I downloaded that library few days ago, I deleted what i have and clone that and now it compile OK!
thanks!

closed 2018-11-10 by luisgcu