xoseperez/espurna #1334: SI71021 with Sonoff TH reports incorrect values

eschava, 2018-11-12:
I have Sonoff TH with SI7021 sensor installed in my bathroom (very often humidity reaches 100%)
With an update to the latest 1.13.3 with build flag -DDHT_SUPPORT=1 (sorry, do not remember what version was installed before) when humidity is very high (before it was reported as 100%) - I receive values less than 10%. You can see examples on the attached chart. Looks like sensor reports values >100 and 100 is just subtracted from the value.

[image]

eschava, 2018-11-13:
One interesting finding: before/after dropping value some intermedial value is received.
Example:
91, 100, 100, 100, **11**, 0, 0, 0, 0, 0, **33**, 100, 100, 100, 98, 96

xoseperez, 2018-12-09:
This is strange since the `_magnitudeProcess` method in the `sensor.ino` file adds a constraint to the humidity value so it will always have a value between 0 and 100.

Not sure where the error could be, it could as well be in the MCU in the sensor. I have one at home, I will try to reproduce the error...

mcspr, 2019-04-06:
So it is an overflow? If it produces some bogus data, filter receives raw unconstrained value:
https://github.com/xoseperez/espurna/blob/903ceb1e844586a6b954c294d7a4438c75a08467/code/espurna/sensor.ino#L1443
https://github.com/xoseperez/espurna/blob/903ceb1e844586a6b954c294d7a4438c75a08467/code/espurna/sensor.ino#L1463
Which is used in the report (and constrained again):
https://github.com/xoseperez/espurna/blob/903ceb1e844586a6b954c294d7a4438c75a08467/code/espurna/sensor.ino#L1506-L1507