xoseperez/espurna #1336: Password issues

mrwee, 2018-11-12:
Having flashed 1.13.3 on an H801 (First flashed a Blank 1MB file, then the Espurna) and setting the Wi-Fi credentials via serial (set wifi.pass0 / set wifi.ssid0 / wifi.reset), I can browse to the H801 via my Wi-Fi. 
At first connect I'm prompted for the standard password, then asked to change it.
After that the module reboots and Espurna keeps complaining when I hit "save":
First: "The password you have entered is not valid, it must be 8..63 characters and have at least 1 lowercase and 1 uppercase / number!"
Secondly: Passwords are different!

Even if I try to to change the password via serial ("set adminPass") or via web, I can't get rid of these two errors. The password I'm trying to set, complies with the complexity rules.

mcspr, 2018-11-13:
Hello.

I remember this being a bug between 1.13.2..1.13.3, but release version should not have it.
Are you using .bin from the releases? Or, if building yourself, update the sources to 1.13.3 / dev

mrwee, 2018-11-13:
I'm using the .bin version. Interestingly enough I haven't experienced it in "espurna-1.13.3-itead-sonoff-dual.bin" which I also flashed yesterday.

ColinShorts, 2018-11-13:
What are you setting the device name as? - not sure if it matters, but I did have issues with longer hyphenated device names in my last round of flashing.

mrwee, 2018-11-14:
I set it as "H801-Espurna1", but I also experienced it with the default "ESPURNA-XXXXXX" hostname. Changing the hostname to "ESPURNA" did not help either.
If I go to "Admin" and change the password, then it doesn't complaint until it's been rebooted.