xoseperez/espurna #1339: Another revision of the Magic Home LED Controller

peterhoeg, 2018-11-13:
I have one with "ZJ-WFMN-A V1.1" printed on it which according to https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-with-ESP8285 is a version with the ESP8285. 

Is this supported by any of the existing profiles?

racerx, 2018-11-17:
I have this same version with A. Was able to get it to work as described in the wiki for the ESP8285 version, but I have not been able to see the white LED turn on. The color works. So I am now going to have to pinout the drive fet for the white chain.

peterhoeg, 2018-11-17:
@racerx, did you comment and then remove the comment again?

racerx, 2018-11-17:
Sorry, I did . I was on the wrong issue. I have this same unit and got it to work using the configuration that I created here: https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-with-ESP8285

I then realized that I was commenting in espurna and not on tasmota so I deleted it.

peterhoeg, 2018-11-20:
This is slightly off-topic @racerx, but how exactly do you mount/hold the pogo pins to the pads instead of soldering?

racerx, 2018-11-20:
Hi Peter,

I am soldering directly to the board with a header and a FTDI TTL-232R3V3 serial adapter. Please know that the signals are 3V3 but VCC on that adapter is 5V and will fry your module if you connect it.

Only need to connect TX,RX and GND to the adapter.
During power up you need to hook IO0 and GND together. I did this with a male and female jumper cable.

[image]
[image]
[image]

peterhoeg, 2018-11-21:
Thank you for the super detailed instructions.

I was hoping that it was possible without actually soldering (I managed to flash a B1 by holding a 5 pin header pin very still) and so the mention of pogo pins got my hopes up but I guess that's not doable without a 3d printer jig.

I'll try this out on the weekend, thanks again!

peterhoeg, 2018-11-21:
It worked and no soldering was required. Crocodile plugs to connect RX, TX and GND (with plastic on the other side of the PCB to ensure there were no shorts), a free wire for IO0 and a male dupont for power (I powered it from the FTDI instead of the normal power plug). Then it was just about holding my hands *very* still while the firmware uploaded.

racerx, 2018-11-21:
Very cool! So you covered the whole other side of board with plastic and then powered 5V to the normal power input with the red wire in my picture?

peterhoeg, 2018-11-22:
A man smarter than me would probably have taken photos to document this while everything was open and connected... 

> you covered the whole other side of board with plastic

Correct, I simply used a small plastic bag folded a few times to cover the entire other side (the side with the chips)

- TX was connected using a crocodile plug from the left (using your photos for direction).
- RX was also connected using a crocodile plug from the left but with a small piece of plastic to cover the pad it had to "reach over"
- GND was connected to the top GND from the right also with a crocodile plug and a small piece of jumper wire used to connect to IO0 (the lower GND was not used at all)
- Power was supplied to the V33 pad using 3.3V from the FTDI

xoseperez, 2018-12-09:
Added support for the A/B variants in the `dev` branch. 
@racerx @peterhoeg Do you mind if I add your flashing instructions to the wiki?

racerx, 2018-12-09:
@xoseprez yes, please add. 

Thanks,
Rex

peterhoeg, 2018-12-10:
@xoseperez  - do you want a wiki PR for the wiki changes? I can write it properly.

Do you think it would be appropriate to make recommendations on the hardware wiki? I just received a H801 and flashed it last night which took about 5 minutes in total - I personally would *never* bother with the Magic Home again and just use H801 instead.

DOSputin, 2018-12-10:
Interesting about the H801.  I'll have to try to come by one of those.

Amazon sent me a `magichome ZJ-WFMN-A V1.1` _by accident_ ( errors = free stuff 🥇)    

Found this issue after trying to flash the other magichome firmware with no joy.

I got the new `ZJ-WFMN-A V1.1` firmware compiled and flashed, and [minicom](https://formulae.brew.sh/formula/minicom) talks to it over a [PL2303-TTL usbserial](http://www.prolific.com.tw/US/ShowProduct.aspx?p_id=229&pcid=41) on Mac OS High Sierra.  Messed about a bit with the config that way, get it set up to my liking. 

Thanks for all the hard work fellows!  

Only problem now, is it seems need to re-config the RGB channels?

[image]

xoseperez, 2018-12-10:
@DOSputin These MagicHome devices are kind of a nightmare. They all have different channel configurations. The ones I set for these models are based on the documentation above in the Tasmota project. What GPIO control what channels in your device?

@peterhoeg It will really help me if you write a PR for the wiki with the docs. Thank you. Yes, some devices are easier to flash than others. Still, the MagicHome are small and use to feature RF/IR which the H801 doesn't.

racerx, 2018-12-10:
ZJ-WFMN-A V1.1

Field	Value	Function
Module type	18 Generic	Module type
D2 GPIO4	51 IRRecv	IR Remote (optional, view console for debugging!)
D1 GPIO5	38 PWM2	BLUE
D6 GPIO12	37 PWM1	RED
D7 GPIO13	39 PWM3	GREEN
D8 GPIO14	40 PWM4	WHITE

racerx, 2018-12-10:
The pinout on 1.1 is different. I added the correct pins to the sonoff wiki here:

https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-with-ESP8285

closed 2018-12-11 by xoseperez

Robert-K, 2019-01-09:
@Floriansimmer Would you mind sharing a compiled .bin with your changes? I can't get it to compile because of some library compatability issues, apparently (using Arduino IDE).

mafyata, 2019-04-15:
Does someone have original firmware file for magic home rgb with ir remote zj wfmn-a v1.1 ?

vsterian, 2019-07-11:
> It worked and no soldering was required. Crocodile plugs to connect RX, TX and GND (with plastic on the other side of the PCB to ensure there were no shorts), a free wire for IO0 and a male dupont [...]

Hello!
I have also a "zj wfmn-a v1.1" and i was able to successfully put a new firmware on it.My problem is that after i download the new software, the ESP8285 does not broadcast any SSID.
I have 3 main questions:
1. The GPIO pin must be shorted with GND only on boot-up?
2.Can you tell me please what firmware did you use in order to successfully use the Controller? 
3.Is there any way of "reseting" the controller, or just the wi-fi module (like on H801 with on/off for 4/5 times)?

Thank you,
Vlad

peterhoeg, 2019-07-11:
> My problem is that after i download the new software, the ESP8285 does not broadcast any SSID.
> I have 3 main questions:
>
> 1. The GPIO pin must be shorted with GND only on boot-up?

Yes, that's at least how I did.

> 2. Can you tell me please what firmware did you use in order to successfully use the Controller?

Just regular espurna, but self-compiled.

> 3. Is there any way of "reseting" the controller, or just the wi-fi module (like on H801 with on/off for 4/5 times)?

Not that I am aware of, no.

sovanyio, 2020-05-04:
Just wanted to chime in that I also have a ZJ_WFMN_A_11 that needed the white GPIO change to 15, are there older devices with GPIO 14?
Should we update the value or create a second config?

mcspr, 2020-05-05:
@sovanyio 2nd config might be better, naming it though...
Since 37f62477728bc8ff621e102286e4cc188310aab4 we can change PWM pin at runtime through `ltDimmerGPIO#` where `#` 0...4 maps to hard-coded `LIGHT_CH#_PIN#` 1...5

sovanyio, 2020-05-07:
Oh nice I didn't see that in the wiki when I was looking--ended up compiling the modification myself. Runtime change :+1: