xoseperez/espurna #1340: Question: 12-bit PWM?

GaelZorro, 2018-11-13:
Hello!
I'm trying to get 12-bit PWM dimming (values 0-4095) on my MagicHome LED controllers, instead of 8-bit (values 0-255)... Just so that I can use powerful LEDs, and yet not get blinded when the family gets up at night.
How would you advise to do that? 
I tried to just change the constants LIGHT_MAX_PWM values and LIGHT_MAX_VALUE to 4095, but I guess there must be more legwork involved, because it doesn't work :-)
I'm not using RGB, HSV, brightness or colorpicker.

GaelZorro, 2018-11-13:
And thanks to you and to the contributors for your great software!

xoseperez, 2018-12-08:
PWM engine depth is defined by LIGHT_MAX_PWM and you can set it to 4095. But internally ESPurna will still use 8-bit values so LIGHT_MAX_VALUE must stay at 255.

Adrianotiger, 2019-01-16:
Is there a way to increase this value?
I have 4 white channels and need to correct the gamma of each channel for a linear dimmer. 255 is not really much, if you want to use a look-up table for example…

I don't know where in the code I can change it from byte to int so that it would be possible to use a bigger number.

xoseperez, 2019-01-20:
Unfortunately, it's not that simple. There are a number of places in the code where a resolution of 8 bits is used (starting with the web UI, for instance).

closed 2019-03-28 by stale[bot]