xoseperez/espurna #1347: Question according SHT3X sensor

Schwenkner, 2018-11-15:
I'm currently adopting the espurna code to "more" C++, since I use it with the Visual Studio IDE instead of Platform.IO. I could adopt alreday a lot of the sensors, including BMX280, which is an I2C based sensor. From that I added a sensor for BMP180, which is also I2C based. This works fine, the BMX280 I didn't try yet.

Yesterday I added the SHT3X sensor and run into a problem. This is the original code:
`

            // cTemp msb, cTemp lsb, cTemp crc, humidity msb, humidity lsb, humidity crc
            _temperature = ((((buffer[0] * 256.0) + buffer[1]) * 175) / 65535.0) - 45;
            _humidity = ((((buffer[3] * 256.0) + buffer[4]) * 100) / 65535.0);
`
Formerly I used my SHT30 sensor with LUA, which has similar code and also worked:
`

            -- calculate temperature and humidity (2 data bytes + 1 checksum byte)
            t = ((((data:byte(1) * 256) + data:byte(2)) * 1750) / 65535) - 450
            h = ((((data:byte(4) * 256) + data:byte(5)) * 1000) / 65535)
`

What I figuered out was, that I have to switch the bytes in the buffer in my code. I have lsb first and then msb. Switching this works fine. The crc is the 3rd byte. But since the BMP180 uses the same routines to read from I2C, I do not expect an error in my adopted I2CSensor class.

Do you have any idea what may be wrong?

mcspr, 2018-11-16:
I am a bit confused, but what is your question regarding ESPurna?

For lua code - if bytes are other way around, it is likely big/little endian mistake.

Schwenkner, 2018-11-17:
I think, you did not understand my post.
In both cases, Espurna and LUA first the order is msb, lsb, crc. But my implementation which is the same as in Espurna, I have the order lsb, msb, crc. But I do not know, how the read function is implemented in LUA. I do not think it is a big/little endian problem, since according to the implementation an byte array of 6 bytes is read, which does not have endian problems.

What I would ask, if there is someone using the ESPurna implementation and getting correct values.

mcspr, 2018-11-18:
So you mean that this part
https://github.com/xoseperez/espurna/blob/d82f9b4e2f809bc194f1a56ad60c6a4085f45841/code/espurna/sensors/SHT3XI2CSensor.h#L71-L75
does not fill the buffer correctly?

e.g. 6 4 crc 6 4 crc is read as 4 6 crc 4 6 crc ?

Schwenkner, 2018-11-19:
Yes, I meant this peace of code. When I change msb and lsb, it works for me.

Could someone verify, that this code works correctly? Then i have to search somewhere else in my code, i.e. in i2c_read_buffer(_address, buffer, 6);

xoseperez, 2018-12-12:
I don't have any SHT3X I2C at hand, but I all the implementations of the protocol I have seen use the first byte as MSB, including the sample code by Sensiron themselves (https://github.com/Sensirion/arduino-sht/blob/master/SHTSensor.cpp).

The datasheet also (of course) specifies that the data is sent MSB-first (https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/0_Datasheets/Humidity/Sensirion_Humidity_Sensors_SHT3x_Datasheet_digital.pdf)

closed 2019-02-17 by stale[bot]