xoseperez/espurna #1349: Cannot connect to MQTT

chenfengfeng, 2018-11-15:
hardware:S20
version:1.13.3
homeAssistant config:
`
mqtt:
  broker: 192.168.31.200
  port: 1883
  client_id: home-assistant
  keepalive: 60
  username: sonoff
  password: raspberry
 
switch:
  - platform: mqtt
    name: "fan"
    state_topic: "sonoff/fan/relay/0"
    command_topic: "sonoff/fan/relay/0/set"
    payload_on: "1"
    payload_off: "0"
    retain: true
    qos: 0
`
192.168.31.200 is my home-assistant IP

espurna config:
[image]

espurna debug:
`
[464359] [MQTT] Connecting to broker at 192.168.31.200:1883
[464362] [MQTT] Connecting as user sonoff
[464363] [MQTT] Client ID: home-assistant
[464366] [MQTT] QoS: 0
[464367] [MQTT] Retain flag: 1
[464368] [MQTT] Keepalive time: 300s
[464370] [MQTT] Will topic: sonoff/fan/status
[464510] [MQTT] Connected!
[464513] [MQTT] Unsubscribing to # (PID 1)
[464515] [MQTT] Subscribing to sonoff/fan/relay/+/set (PID 2)
[464516] [MQTT] Subscribing to sonoff/fan/pulse/+/set (PID 3)
[464518] [MQTT] Subscribing to sonoff/fan/led/+/set (PID 4)
[464523] [MQTT] Subscribing to sonoff/fan/action/set (PID 5)
[464530] [MQTT] Sending sonoff/fan/app => ESPURNA (PID 1)
[464535] [MQTT] Sending sonoff/fan/version => 1.13.3 (PID 1)
[464542] [MQTT] Sending sonoff/fan/board => ITEAD_S20 (PID 1)
[464545] [MQTT] Sending sonoff/fan/host => fan (PID 1)
[464551] [MQTT] Sending sonoff/fan/ip => 192.168.31.201 (PID 1)
[464555] [MQTT] Sending sonoff/fan/mac => 5C:CF:7F:E9:76:91 (PID 1)
[464562] [MQTT] Sending sonoff/fan/rssi => -56 (PID 1)
[464567] [MQTT] Sending sonoff/fan/uptime => 174464 (PID 1)
[464571] [MQTT] Sending sonoff/fan/datetime => 2018-11-15 23:11:30 (PID 1)
[464578] [MQTT] Sending sonoff/fan/freeheap => 19536 (PID 1)
[464584] [MQTT] Sending sonoff/fan/relay/0 => 1 (PID 1)
[464588] [MQTT] Sending sonoff/fan/vcc => 3212 (PID 1)
[464594] [MQTT] Sending sonoff/fan/status => 1 (PID 1)
[464597] [MQTT] Sending sonoff/fan/loadavg => 3 (PID 1)
[464603] [MQTT] Subscribe ACK for PID 2
[464605] [MQTT] Received sonoff/fan/relay/0/set => 1 - SKIPPED
[464610] [MQTT] Subscribe ACK for PID 3
[464613] [MQTT] Subscribe ACK for PID 4
[464617] [MQTT] Subscribe ACK for PID 5
[465638] [MQTT] TCP Disconnected
[465641] [MQTT] Disconnected!
`

Wisnaeme, 2018-11-28:
I'm in exactly this situation too. It's clearly connected and ACKing but the fields aren't updating.

Sonoff basic 1.13.3 for me.

mcspr, 2018-12-01:
What broker is it? hass.io mosquitto?
To rule out credentials issue - does something like `mosquitto_sub` / `mosquitto_pub` work with the same ones?

Wisnaeme, 2018-12-02:
Hi, I got to the bottom of this one that I also posted over here... it was a Home Assistant issue.

https://community.home-assistant.io/t/espurna-setup-going-crazy-here/80864/4

"OK. Update. The internal MQTT broker was not responding properly. As I’m so new to this I was trusting it to be working. I moved over to using Windows Mosquitto as a broker and it worked fine straight away.
As I use an Unraid server for MQTT, I then moved to a different MQTT broker as a Docker and that worked too. I’ve no idea why the internal docker was not working. Thanks for reading!"

Everything has worked great since, although I've had a couple of unexpected lockups on the BasicDHT prebuilt code while it was sending Thingspeak data and MQTT. I'm still researching that separately.

Thanks for coming along and helping!

chenfengfeng, 2018-12-02:
@Wisnaeme So how do you currently connect to MQTT?