xoseperez/espurna #1350: Support Blitzwolf BW-SHP2 v2.3

J0s3f, 2018-11-15:
I bought a Blitzwolf SHP2 and got a new Hardware revosion which looks exactly like the on described here: https://github.com/arendst/Sonoff-Tasmota/wiki/Gosund-SP1#new-hardware-revision

The espurna-1.13.3-blitzwolf-bwshp2.bin files boots and WiFi works, but all sensor values are either missing or wrong.

It works with Tasmota, but I would love to use espurna.

There are the PIN/config settings they are using: https://github.com/arendst/Sonoff-Tasmota/issues/4297#issuecomment-436804860

Valcob, 2018-11-15:
@J0s3f Hi, try this 
`
#elif defined(BLITZWOLF_BWSHP2V2.3)

    // Info
    #define MANUFACTURER_DEF            "BLITZWOLF"
    #define DEVICE_DEF                  "BWSHP2V2.3"

    // Buttons
    #define BUTTON1_PIN                 3
    #define BUTTON1_MODE                BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY               1

    // Relays
    #define RELAY1_PIN                  14
    #define RELAY1_TYPE                 RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN                    1
    #define LED1_PIN_INVERSE            1
    #define LED2_PIN                    13
    #define LED2_PIN_INVERSE            1
    #define LED2_MODE                   LED_MODE_FINDME
    #define LED2_RELAY                  1   // maybe no ? move it to led 1?

    // HJL01 / BL0937
    #ifndef HLW8012_SUPPORT
    #define HLW8012_SUPPORT             1
    #endif
    #define HLW8012_SEL_PIN             12
    #define HLW8012_CF1_PIN             5
    #define HLW8012_CF_PIN              4

    #define HLW8012_SEL_CURRENT         LOW
    #define HLW8012_CURRENT_RATIO       25740
    #define HLW8012_VOLTAGE_RATIO       313400
    #define HLW8012_POWER_RATIO         3414290
    #define HLW8012_INTERRUPT_ON        FALLING
`

tijnema, 2018-11-16:
I can confirm above settings are correct. (And created pull request #1351 for this)

J0s3f, 2018-11-17:
I had time to compile and reflash today and can also confirm it works.
Thanks a lot!

supercharley, 2018-11-21:
Would someone send me the finished binary file?
Unfortunately I can not make it! I need it!
e-mail: supercharley@gmail.com
Thank you very much!

rradar, 2018-12-20:
Question: Can I flash "espurna-1.13.3-blitzwolf-bwshp2.bin" on my device (v2.3) will it work?