xoseperez/espurna #1353: Sonof Basic Add Network

spenjt, 2018-11-18:
I have just added a second router to make connections to my Sonoffs better. I want 2 SSID added to the Sonoffs so that it can be configured using one of the routers and then moved later.
When adding a second network SSID it is changing the first SSID to match the second and therefore there is only one SSID that can be used.

spenjt, 2018-11-18:
Sorry, this is with a Sonoff Basic with the most recent version 1.13.3

spenjt, 2018-11-18:
I have made the SSIDs separate so that our phones don't connect to the other router. It's just that the after configuring Espurna with two separate SSIDs and rebooting., both SSID entries have the same SSID.

xoseperez, 2018-12-12:
Do you mean you stored two different SSIDs in ESPurna and, after rebooting, they are not the same?

spenjt, 2018-12-12:
As I understand it, you can enter multiple WIFI connections and it picks the strongest connection. I had entered details for two different routers (i.e. two different SSIDs). When I rebooted  one of the SSIDs had been renamed, so that both entries had the same SSID.

spenjt, 2018-12-12:
It's not important to me at all. Just thought I would make you aware of it.

xoseperez, 2018-12-13:
Does it always happen? Can you reproduce the issue (configure two different SSID, reboot and check the configuration)?

closed 2019-02-18 by stale[bot]