xoseperez/espurna #1354: Assign names to sensors

Misiu, 2018-11-18:
Hi,
currently, when You have multiple DS18B20 sensors, the temperature is sent to MQTT as described in [this issue](https://github.com/xoseperez/espurna/issues/377)
like this:
```
ESPURNA_343624/temperature/28FF732970160440 => 17.0
```
or like this:
```
ESPURNA_343624/temperature/0 => 17.0
ESPURNA_343624/temperature/0/address => 28FF732970160440
```
or like this:
```
ESPURNA_343624\data => {'temperature/0': 17.0, 'temperature/0/address': '28FF732970160440', ...}
```

It would be better if it would be possible to name each sensor via Web UI so the message send to MQTT could look like this:
```
ESPURNA_343624/temperature/kitchen_floor => 17.0
```

This option would allow easy configuration of additional sensors.
Imagine You try to add 5 DS18B20 sensors.
Currently when You plug all of them You must check each ony by hand and modify code in OpenHab to assign address with variable.

With ability to assign name to each sensor You could connect single sensor, assign name to it via WebUI, connect second one, rescan sensors via WebUI and You will notice which one in old sensor (has name assigned) and which is new (no name).

| Sensor address  |      Friendly name      |  Assign |
|----------|:-------------:|------:|
| 28 FF 73 29 70 16 04 40 |  kitchen_floor | Assign |
| 28 FF 73 2B 45 4C 00 10 |    kitchen_outside_window   |   Assign |
| 28 FF 2D 25 2B 2D 12 BB |  |    Assign |

This way code in places that use MQTT messages would be easier to write, because You would be able to use friendly name instead of addresses.

Another big adventage of this solution would be the ability to simple replace broken sensor.
You uplug old one, plug new one and in WebUI You assigne name that old one used. Signe change in ESPurna and no changes in other places.

Gaff, 2018-12-17:
This also applies to switches. I can name the host but I don't necessarily want to call my switches `heating_1` and `heating_2`!

For users of home assistant MQTT discovery there's a relatively simple solution -[ instead of setting the name to be the hostname
](https://github.com/xoseperez/espurna/blob/59685150b4ea6358253a24c4ba6792cf2e695131/code/espurna/homeassistant.ino#L76), we could make it configurable via the GUI. Then the discovery message would contain a nice name for each switch / sensor, but the MQTT topics could remain exactly as they are.

excalq, 2023-05-28:
Very much looking forward to this for my Sonoff 4ch-pro powered irrigation control!