xoseperez/espurna #1356: Backup & Restore to original firmware on Sonoff Pow R2 is not working

HobbytronicsPK, 2018-11-19:
Hello,
I have backed up the firmware according to instructions given under your [wiki article](https://github.com/xoseperez/espurna/wiki/Backup) and another article [Backup & Restore Official Firmware of Sonoff WiFi Smart Switches](https://hobbytronics.pk/sonoff-original-firmware-backup-restore/) on my Sonoff Pow and Sonoff Pow R2 using the command as given below.

`esptool.py --port COM5 read_flash 0x00000 0x100000 image1M.bin`

Restored worked on Sonoff Pow but it does not work on Sonoff Pow R2. It gets restored successfully but there is no LED blinks observed after restore. Here is command used for restore purpose.

`esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x0 image1M.bin`

I also erased the flash prior restoration. 

`esptool.py --port COM5 erase_flash`

If I re-install your pre-built Espurna binary, it starts working again. Only the restore of original firmware does not work. I have tried it on different Sonoff Pow R2 devices without any success. But on Sonoff Pow (which is Version 2, not R2) it always works as expected.

mcspr, 2018-11-22:
Hi,

I had noticed that R2 wiki mentions it using 4MiB flash. Given the instructions are for 1MB boards, maybe backup is incomplete? And I did notice you asked the same at Tasmota repo, where they have similar guess.

HobbytronicsPK, 2019-01-17:
yes. it was the issue with flash size. i have updated the instructions in **[Backup & Restore Official Firmware of Sonoff WiFi Smart Switches](https://hobbytronics.pk/sonoff-original-firmware-backup-restore/)**

closed 2019-01-17 by HobbytronicsPK

evlo, 2019-04-07:
Can you please share backup image for R2?

HobbytronicsPK, 2019-04-30:
@evlo have a look on to [Official Sonoff Firmware Binary Files](https://blog.hobbytronics.pk/sonoff-original-firmware-backup-restore/#Official-Sonoff-Firmware-Binary-Files).

evlo, 2019-05-07:
i was never able to flash original firmware back (or even get it from r2 i purchased for that purpose), with your bin I do get success flashing it like this esptool.exe -vv -cb 115200 -cp com3 -ca 0x00000  -bz 4M -bm dout -cf c:\sonoff\Sonoff_Pow_R2_4M.bin
but my r2 does not boot up with it

then if i go for espurna like
esptool.exe -vv -cb 115200 -cp com3 -ca 0x00000  -bz 1M -bm dout -cf c:\sonoff\sonoff.bin 
success on first try
i'm using esptool-ck as the python gives me weird codes even for espurna, while with ck it works and is not currupted, don't know why, maybe python3, maybe my flashing tool, which is arduino mega, as i fried r3 attempting to make it ftdi only :), maybe windows, but i just gave up

mcspr, 2019-05-07:
@evlo have you tried different flash modes? you are setting DOUT, but original firmware is QIO:
```
$ xxd Sonoff_Pow_R2_4M.bin  | head -1
00000000: e903 0040 7c05 1040 0000 1040 200a 0000  ...@|..@...@ ...
$ xxd espurna-1.13.5-itead-sonoff-basic.bin | head -1
00000000: e901 0320 9cf2 1040 00f0 1040 6805 0000  ... ...@...@h...
```

Flash mode values:
https://github.com/esp8266/Arduino/blob/e67cc90b7a4914c6c3a6fd795183fefc5e910094/cores/esp8266/Esp.h#L124

We are looking at the 3rd byte:
https://github.com/esp8266/Arduino/blob/e67cc90b7a4914c6c3a6fd795183fefc5e910094/cores/esp8266/Esp.cpp#L308

evlo, 2019-05-07:
That is too hardcore for me, I don't understand that deeply what is going on in serial programming, however I tried `esptool.exe -vv -cb 115200 -cp com4 -ca 0x00000  -bz 4M -bm QIO -cf c:\sonoff\Sonoff_Pow_R2_4M_.bin` and still offical firmware did not boot up (when esptool.py works and starts flashing firmware it also does not boot put, but i did not tried QIO with it)

mcspr, 2019-05-07:
This issue https://github.com/espressif/esptool/issues/418 goes through some possible problems when flashing. I'd look at power situation and maybe wiring. Consider using different 3v3 power supply?

Note that esptool.py is preferable option when flashing, as it is official espressif tool (and it much more stable and faster that reverse engineered -ck)

mauleonjr, 2019-09-04:
I did everything correctly and I already copied the fimware of a new device to the device to recover successfully. BUT when I try to use the eWelink application and try to add the device (it says in a state of synchronization that a second generation device has been found), an error box pops up in the app saying: Invalid Data. And the operation is canceled

Now what will be the problem?